fieldOfView/Cura-OctoPrintPlugin #2: Head tries to move too far right

nicolinux, 2016-09-05:
When I click on "Print with OctoPrint" the print head instantly moves to the back right and tries to go as far as possible producing a horrid rattling noise.
If I start the print (of the .gcode file that has been produced by Cura Desktop with the OctoPrint plugin), the prints starts normally but the nozzle is not heated up...

Tested with Cura Desktop 2.3.0-BETA on the Mac and OctoPrint 1.2.15.

fieldOfView, 2016-09-05:
What printer?

For the UM2, be sure to use the Machine Settings to set the gcode flavor to Reprap.

nicolinux, 2016-09-05:
UM2+
Where do you access the Machine Settings in Cura 2.3? This was available in Cura 15.x but not the new Cura. Is it possible now to set the gcode flavor via the gui or do you need to edit the cura definitions files?

fieldOfView, 2016-09-05:
For the UM2+ you need to edit the definition files unfortunately. I'll try to get that fixed before the 2.3 final release.

nicolinux, 2016-09-05:
Thanks, that would be great. You could also mention it on the front page here since not many people are aware of it and might run into the same issue.

closed 2016-09-05 by nicolinux