fieldOfView/Cura-OctoPrintPlugin #6: Add Printer without autodiscovery

larp-welt, 2016-09-19:
Hi!

For some unknown reason, the autodiscovery does not work for me! No great thing...

But is there a way to connect to a printer without autodiscovery? 

Bye

fieldOfView, 2016-09-19:
At the moment the autodiscovery is required for the function of the plugin. I have been meaning to add a manual address as well as port selection. Hopefully I'll manage to do that before 2.3 final is released.

agg23, 2016-09-24:
@larp-welt In the meantime, you can easily manually add your printer by inserting the following line on line 26 of OctoPrintOutputDevicePlugin.py.  Replace Printer Name with what you want the printer to be called and 127.0.0.1 with your printer's address.  When you open Cura, your printer should show up in the autodiscovery list.

`self.addInstance("Printer Name", "127.0.0.1", None)`

fieldOfView, 2016-09-25:
Thanks a lot for that suggestion; doing it like that sparked a much simpler idea than what I had in mind for manual selection

agg23, 2016-09-25:
@fieldOfView I was going to add the manual selection and submit a pull request, but dealing with PyQt proved to be more than I wanted to mess with.

fieldOfView, 2016-09-27:
It is now possible to add a manual instance without hacking the code. Instead you have to hack the preference file. Add the following lines to cura.cfg (editing the parameters to suit your setup):
```
[octoprint]
manual_instances = [{"name": "OctoPrint instance", "address": "octopi.local", "port": 80, "path": "/"}]
```

A proper interface for this is in the works. I'm leaving this issue open until I've had time to add it.

fieldOfView, 2016-10-05:
There is now a UI to add/edit/remove manual instances.
Note that the preference format has changed subtly, and this may cause issue if you have already added an instance to the preference file:
```
[octoprint]
manual_instances = {"OctoPrint instance": {"address": "octopi.local", "port": 80, "path": "/"}}
```

closed 2016-10-05 by fieldOfView

oxcxsxixbxaxn, 2016-10-05:
Still doesn't work for me...  
 I have installed the latest patches but Discovery just does not work on Windows 7 and so I don't see any octoprint settings, only machine settings. To enable printing before the patch I copied the manual_instances file from a windows 10 machine. the windows 7 machine can now print to octoprint but still no octoprint settings

I have tried clearing our the manual_instances and adding the line you suggest but still not showing anything but machine settings...

fieldOfView, 2016-10-06:
Please create a new issue, and include a pastebin of your logs.

filipgoc, 2017-04-18:
Just want to mention my use case : I like to use the octopi.local version instead of direct IP because then my password manager automatically logs me in. Small thing, but saves a couple of clicks, and thus improves the UX. Cheers.

fieldOfView, 2017-04-18:
@filipgoc: this issue has long been closed. It is already possible to add instances manually, which can take ```octopi.local``` as an address. If this is not working for you, please create another issue.

filipgoc, 2017-04-18:
@fieldOfView sorry for the confusion. The way it works now is PERFECT for me. I just wanted to illustrate a use case for some people that may come here later... an inspiration so to say.

Maybe commenting on a long closed thread is not a way to do this. Or maybe I have to be more explicit next time as in: *I don't request anything. All is well.*