fieldOfView/Cura-OctoPrintPlugin #7: Feature Request: Upload to OctoPrint without starting to print

larp-welt, 2016-09-19:
Hi!

I'm missing a function to send my gcode to OctoPrint without starting to print immediately.

I've just hacked that for me by changing the line:

` ## self._post_part.setBody(b"true")`
`self._post_part.setBody(b"false")`

But this way, it never starts printing! ;-)

Bye

fieldOfView, 2016-09-19:
I think I will implement this as a second option in the output selection (botom right). The first option would be "Print with OctoPrint" (selected by default) and the second would be "Send to OctoPrint". Once you have sent a file to octoprint, it would be up to you to start the print.

fieldOfView, 2016-09-27:
Fixed by https://github.com/fieldOfView/OctoPrintPlugin/commit/a039fabb65ce47d79a8a822e88e3e6679229db06
Implementation is slightly different. There is now an option to automatically print on the Connect to OctoPrint dialog. This is a per-machine option.

oxcxsxixbxaxn, 2016-10-05:
How would I manually enable this option to send only?  I have installed the latest patches but Discovery just does not work on Windows 7 and so I don't see any octoprint settings, only machine settings.  To enable printing I copied the manual_instances file from a windows 10 machine.  the windows 7 machine can now print to octoprint but still no octoprint settings

fieldOfView, 2016-10-06:
You should not have to manually enable this option, because there is a UI to do this. Discovery not working has nothing to do with there being a "Connect OctoPrint" button on the "Manage Printers" preference page. If you do not have that button, please create a new issue and include a pastebin of your log.

oxcxsxixbxaxn, 2016-10-06:
I backed up and removed the AppData\Local\cura directory to capture logs, restarted cura and now the buttons show up.  So I have added the printer manually connected left the "automatically start print" option unchecked.  It starts the print straight away...   I went back in and checked the box to automatically start the print and tried again to see it something changed, it didn't and when I went back to uncheck the option again, it was already unchecked, seems this option does not get stored.

fieldOfView, 2016-10-06:
I'll look into it. I must admit I have not tested this feature very much, because I always just want to start the print ;-)

fieldOfView, 2016-10-12:
This should now be fixed. @oxcxsxixbxaxn, could you try again?

oxcxsxixbxaxn, 2016-10-16:
Do I need to reset the config or something?  I shutdown Cura, copied over the latest checkin and started up Cura.  The tickbox for for not printing straight away if not ticket yet it prints straight away.  I tick the box, close, and go back in and it's still unticked...

[image]

A further request related to this is whether Octoprint exposes the ability to send a gcode to the printers SD card and print from there?

fieldOfView, 2016-10-17:
Does the checkbox at least stay checked now if you check it and reopen the window?

fieldOfView, 2016-10-17:
@oxcxsxixbxaxn: It works for me now. The checkbox state is properly stored, and when the checkbox is not checked the gcode is sent to OctoPrint but OctoPrint does not start printing. Instead I get a message that the file has been stored, with a button to open OctoPrint to start printing when it so pleases me:
[image]

Further requests go in separate issues.

oxcxsxixbxaxn, 2016-10-25:
So, this works now, in that the configuration changes between true and false... this is written to the config file.  The behaviour changes according to that selection too.  The only thing that still does not work is the UI.  the tickbox remains unticked, no matter what state it is in, as in the screnshot above.  I can live with that...

Thanks

closed 2016-10-27 by fieldOfView