fieldOfView/Cura-OctoPrintPlugin #10: Warn user about incompatibility with UltiGCode

fieldOfView, 2016-10-05:
Printer in the Ultimaker2 family use UltiGCode flavored gcode. Due to a bug in the firmware, UltiGCode and USB printing don't go well together, so neither does UltiGCode and OctoPrint. The plugin should warn the user about this when it detects the gcode if UltiGCode flavored.

williammck, 2016-10-22:
Not sure how this would/could be implemented in the plugin, but a way I fixed this was by editing `C:\Users\<username>\AppData\Local\cura\user\Ultimaker+2%2B_current_settings.inst.cfg` (will be under a different path for macOS and Linux, [see here](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations)) and adding `machine_gcode_flavor = RepRap (Marlin/Sprinter)` under `[values]`.

fieldOfView, 2016-10-22:
The disadvantage of putting this value in the current_settings is that it is very easily overwritten (when switching materials or qualities and choosing not to keep your current settings). I have added some notes to the readme, with links to alternative machine definitions for the UM2+ and UM2Extended+:

https://github.com/fieldOfView/OctoPrintPlugin/blob/master/README.md

closed 2016-10-27 by fieldOfView