fieldOfView/Cura-OctoPrintPlugin #15: How to install on OSX?

darrelhunt, 2016-11-23:
I go to:
Applications --> Control+Click on Cura --> Show Package Contents --> Resources --> Plugins

This is where it's confusing
That Plugins folder is empty
But there is a Plugins folder within it that has all the standard Cura Plugins
I put the OctoPrintPlugin in that folder and start Cura
It doesn't find it

I even tried putting it in the empty Plugins folder and nothing
Any ideas?

nallath, 2016-11-23:
Start Cura, go to help-> show configuration folder. There should be a plugins folder there.

fieldOfView, 2016-11-23:
What @nallath said.

For completeness, it should also work in the Cura.app/Resources/Plugins folder, or the Cura.app/Resources/Plugins/Plugins folder (nb: the existence of the latter is a bug in the packaging of the application).

If you still cannot get it to work, please put the Cura.log (which you can find in the configuration folder) in a pastebin and post a link here. There may be some log output that is helpful.

darrelhunt, 2016-11-23:
I've already tried
Cura.app/Resources/Plugins
Cura.app/Resources/Plugins/Plugins

When I go to Help --> Show Configuration Folder, it says here

/Users/darrel/Library/Application Support/cura/plugins
I put it there and it still doesn't show "Connect to Octoprint" under Manage Printers

fieldOfView, 2016-11-23:
What sort of printer are you using octoprint with?

darrelhunt, 2016-11-23:
It's a Custom FDM that I've named Prusa Mendel i2

Here's the pastebin

http://pastebin.com/3k9K03he

fieldOfView, 2016-11-23:
It seems the folder inside plugins is named "OctoPrintPlugin.3". The folder must be named "OctoPrintPlugin".

darrelhunt, 2016-11-23:
Wierd, looking at it under OSX GUI it looked right
But in a terminal it was in fact named OctoPrintPlugin.3
Renamed it in terminal and it works

Thanks for the help
Might I suggest adding how to find where to put the plugin directory for OSX in the README?

fieldOfView, 2016-11-23:
Good point

closed 2016-11-23 by fieldOfView