fieldOfView/Cura-OctoPrintPlugin #16: HTTPS (SSL/TLS) support?

computergeek125, 2016-12-06:
Currently, we have an OctoPrint device sitting behind a firewall accessible via an Nginx reverse proxy with TLS enabled and HTTP forwarding to HTTPS for security.  The URL is in the format https://octoprint.example.com/01/, and the camera is available on the same URL

I would like to help update this plugin to support HTTPS, but I can't seem to find any documentation about Cura 2 plugins.  Where did you find your documentation?

fieldOfView, 2016-12-06:
Adding a check for HTTPS to the "Manually added OctoPrint instance" dialog should be fairly straightforward. I'll see if I can add that somewhere this week.

The camera url is currently hardcoded, and would take more work to fix.  I would like to fix that as part of https://github.com/fieldOfView/OctoPrintPlugin/issues/12, and get the URL from OctoPrint so you don't have to configure it twice.

About the plugin documentation... There is very little. I learned by example, and the OctoPrintPlugin is a fork of the UM3NetworkPlugin. But I was also contracted by Ultimaker to work on Cura for a year, so I am partly to blame for the lack of plugin documentation.

computergeek125, 2016-12-06:
Fair enough.  I just found Ultimaker/Uranium so I have more to go off of to reverse engineer things now.

I'm currently working to see if I can get it working with text-based configuration in Cura.cfg.

fieldOfView, 2016-12-06:
If you want to hack in the use of HTTPS, see https://github.com/fieldOfView/OctoPrintPlugin/blob/2.3/OctoPrintOutputDevice.py#L44. The hardcoded camera url is a couple of lines down

computergeek125, 2016-12-06:
The hack sort of worked.  I found a few of errant byte arrays in  [OctoPrintOutputDevicePlugin.py#L63](https://github.com/fieldOfView/OctoPrintPlugin/blob/2.3/OctoPrintOutputDevicePlugin.py#L63), plus another a few lines down (see #17).  This was causing the system to automatically set the path of the OctoPrint server to "/", the default value, since it couldn't find the key (b'path' != 'path').

[image]

That's with hardcoding the https:// and fixing the bytearrays.  I can press a few of the buttons and read the extruder temp, but I see a lot of spam on the log file for `CRITICAL - /.../CrashHandler.py (show [24]):   File "/.../cura/plugins/OctoPrintPlugin/OctoPrintOutputDevice.py", line 491, in _onRequestFinished`  I looked up that line number, and it referrs to acquiring the heated bed temperature of the printer (of which this printer has none).  Not sure if that's interfering with anything yet.

Also, with relation to the "print to octoprint" button: it says that it uploaded and is ready to print, but the web interface of OctoPrint disagrees, and indicates that no new files were uploaded.

fieldOfView, 2016-12-07:
@computergeek125, could you test this branch/pr:
https://github.com/fieldOfView/OctoPrintPlugin/pull/19
It adds a checkbox to use https instead of http to the manual instance dialog.

closed 2017-01-09 by fieldOfView