fieldOfView/Cura-OctoPrintPlugin #21: bed_temperature exception

kontakt, 2017-01-08:
Using version 2.3, the cura log is mostly full of this:

````2017-01-08 04:03:37,910 - CRITICAL - cura.CrashHandler.show [35]:     bed_temperature = json_data["temperature"]["bed"]["actual"]
2017-01-08 04:03:37,912 - CRITICAL - cura.CrashHandler.show [35]: KeyError: 'bed'
2017-01-08 04:03:39,927 - CRITICAL - cura.CrashHandler.show [32]: An uncaught exception has occurred!
2017-01-08 04:03:39,929 - CRITICAL - cura.CrashHandler.show [35]: Traceback (most recent call last):
2017-01-08 04:03:39,931 - CRITICAL - cura.CrashHandler.show [35]:   File "C:\Program Files\Cura 2.4\plugins\OctoPrintPlugin\OctoPrintOutputDevice.py", line 489, in _onRequestFinished

fieldOfView, 2017-01-08:
Thanks for the report. I'll fix it for both the 2.3 and 2.4 release shortly.
For the record, does your printer have a heated bed?

computergeek125, 2017-01-08:
That's the bug I thought might have been system-specific in #19.  Our printer has a separate heated bed control unit that OctoPrint can't see.

I had a simple patch for this back when I was working on the HTTPS branch, but I never pushed it to Github at the time.  It's at computergeek125/OctoPrintPlugin at branch [heated_bed](https://github.com/computergeek125/OctoPrintPlugin/tree/heated_bed), which is based on the 2.4 branch of this repo since I needed HTTPS.  It was my quick 5-minute fix to get Cura to initialize without having the heated bed installed.  

Since the default value for _bed_temperature is 0 in the original PrinterOutputPlugin in Cura, I had it use that value if it couldn't find a bed temperature.

kontakt, 2017-01-08:
I do not have a heated bed.

fieldOfView, 2017-01-09:
This is fixed in both the 2.3 and 2,4 branches. Thanks @computergeek125

closed 2017-01-09 by fieldOfView