fieldOfView/Cura-OctoPrintPlugin #22: OctoPrintPlugin appears in list of plugins, but nowhere else

KTheMan, 2017-01-16:
As the title suggests, it's recognized by Cura 2.3.0, but its presence is nowhere to be found otherwise. It's not where managing printers are and, consequently, not an option to save to.
Running Windows 10 x64 and I've made sure its placement and name are correct as well as having pulled the latest revision. I did at one point have an older version show up but it was not functional and that was a good several months ago.
Any help is appreciated!

KTheMan, 2017-01-16:
I reinstalled my Cura for the third or fourth time and it finally decided to work. No idea what happened.

closed 2017-01-16 by KTheMan

fieldOfView, 2017-01-16:
I this happens again, make sure to post your logs so I can look at the cause.

Something that can cause this is the name of the OctoPrintPlugin folder inside the plugins folder. It *must* be ```OctoPrintPlugin```, not something like OctoPrintPlugin-master.