fieldOfView/Cura-OctoPrintPlugin #25: sent data

jon012198, 2017-02-25:
so it said sending data but none thing happened on the octroprint side  i am useing cura 2.4 on the branch of 2.4

fieldOfView, 2017-02-25:
In the top-right corner of the cura window, there should be a small overlay over the "print monitor" tab/button. It should be green before you try printing. Is it grey for you?

Did you enter the octoprint api key?

jon012198, 2017-02-25:
green and key is there it says connected

fieldOfView, 2017-02-25:
Ok, that's weird. 
Did you select "Store gcode on sd card" or deselect "automatically start print job after uploading"?

fieldOfView, 2017-02-25:
Perhaps your stderr.log can tell me something. Could you post that to pastebin and post the link here?

AlexAhrens, 2017-02-25:
Had a similar problem a couple of days ago. A fresh copy solved the problem.

fieldOfView, 2017-02-25:
A fresh copy of what? Cura, OctoPrintPlugin, or OctoPrint?

AlexAhrens, 2017-02-25:
Sorry, was in a hurry. A fresh copy of the Plugin downloaded from github. I've deleted the existing one in the Plugin-Folder, started cuts and closed it right away. After that copied the new plugin and gave both the printer and octopi a restart.

jon012198, 2017-02-25:
[cura.zip](https://github.com/fieldOfView/OctoPrintPlugin/files/801347/cura.zip)

jon012198, 2017-02-25:
found the problem if you select store on sd card it dose not work  but once it is un selected it works

fieldOfView, 2017-02-25:
As far as i know, save to SD can take an extremely long time, almost as long as printing the same gcode file; the data must go through the same slow usb serial port. It is not recommended to use that option (perhaps i should remove it)

closed 2017-02-25 by fieldOfView

jon012198, 2017-02-25:
yah i would get rid of it since on the facebook group i am in people are just using 64gb sd card on the pi so they dont use a sd card on the control board