fieldOfView/Cura-OctoPrintPlugin #27: Does not get loaded from configuration folder

probonopd, 2017-03-04:
> Download or clone the repository into (...) the plugins folder inside the configuration folder. The configuration folder can be found via Help -> Show Configuration Folder inside Cura. 

It is not loading from there.

```
me@host:~$ find /home/me/.config/cura/
/home/me/.config/cura/
/home/me/.config/cura/cura.cfg
/home/me/.config/cura/plugins
/home/me/.config/cura/plugins/OctoPrintPlugin
/home/me/.config/cura/plugins/OctoPrintPlugin/__init__.py
/home/me/.config/cura/plugins/OctoPrintPlugin/README.md
(...)
```

fieldOfView, 2017-03-04:
I can't help you without logs.
Did you create the plugins folder, or was it already there?

probonopd, 2017-03-04:
I created it

fieldOfView, 2017-03-04:
Then it probably needs to be in another folder on Linux/your distro. Not sure which, sorry

fieldOfView, 2017-03-05:
On Linux you're supposed to use .local/share/cura/plugins

closed 2017-03-05 by fieldOfView

probonopd, 2017-03-05:
Thanks, but please reopen and change title to "Menu entry opens wrong Configuration Folder".

"Help -> Show Configuration Folder" should open `$HOME/.local/share/cura/` in this case.

fieldOfView, 2017-03-05:
No, that is not an issue with this plugin, that's an issue with Cura.

Technically, Cura does open a configuration folder (the folder with cura.cfg in it), but on linux there are 3(!) different configuration folders.

probonopd, 2017-03-05:
Sorry I realized that I confused the tickets. Meant https://github.com/Ultimaker/Cura/issues/1490