fieldOfView/Cura-OctoPrintPlugin #28: Change name before send it and select destination folder

MiguelAngelLV, 2017-03-08:
Would be interesting can set name and folder when upload the file.

fieldOfView, 2017-03-08:
You can already set the name. The name that is used is shown in the lower-right corner of the viewport area (above the print time and amount estimates). Click the name there, and you can edit it.

There currently is no way to select the destination folder.

fieldOfView, 2020-04-04:
Here's a sketch by reddit user /u/Dweller:
[image]
src: https://old.reddit.com/r/octoprint/comments/fus9rz/best_slicer_for_octo/fmemp92/

ChrisNava, 2020-04-27:
GUI Suggestion: Assuming it's possible to get a list of existing folders from Octoprint, just have the "Print with OctoPrint" drop-down menu list the folders under "Save to File" and perform a save when selected.  That way no GUI window change is required.

fieldOfView, 2020-04-27:
I could see that work if a user had only one or two subfolders, but having - say - ten to twenty subfolders would get you a lot of "Print with OctoPrint to folder ...` items in that list. It would also further muddy the functionality of that split button. 

I think I'll probably do it something like this:

[image]

The button will dropdown a menu of available folders, and a future version may or may not have a function to create a new folder. I don't like that the "preview" button is between the print button and the folder selection dropdown button, but sometimes you have to compromise.

fieldOfView, 2020-06-15:
I'm taking a different path, and am implementing this as part of the [Custom Job Prefix](/fieldofview/Cura-CustomJobPrefix) plugin.

fieldOfView, 2020-06-15:
Setting a path is now implemented in the master branches of the OctoPrint Connection plugin and the Custom Job Prefix plugin.

closed 2020-07-16 by fieldOfView