fieldOfView/Cura-OctoPrintPlugin #29: Camera orientation wrong

netlands, 2017-03-13:
If the camera is installed upside down, the orientation can be changed in OctoPrint settings (Flip webcam) so it displays correctly in the OctoPrint web interface.
The camera view in the plugin (Show webcam image) shows the camera upside down (the actual physical orientation).

fieldOfView, 2017-03-13:
This is going to require a change in Cura for actually transforming the webcam image at the time it is being displayed, unless we accept a rather hefty performance hit transforming each frame as it comes in.

Also relies on #12 for getting the configuration from octoprint.

fieldOfView, 2017-06-12:
Fixed in master, which will be released with Cura 2.7

fieldOfView, 2017-06-12:
In Cura 2.7, the plugin gets the url and the orientation for the camera stream from the octoprint preferences.

fieldOfView, 2017-08-02:
A beta of Cura 2.7 is now available. I would appreciate if you could test this functionality

netlands, 2017-08-02:
Tested on beta of Cura 2.7.0. The camera orientation is now correctly displayed as set in OctoPrint.

New issue: because the camera view panel/layer is displayed at maximum (?) size and does not resize when resizing the Cura dialog in Cura 2.7.0 beta, the camera panel obscures part of the Cura UI (including the menu bar) when the dialog is smaller than the camera view. This might be due to the change in overall layout of the Cura interface.

fieldOfView, 2017-08-03:
I'll see if I can fix before the final 2.7 ships

fieldOfView, 2017-08-03:
Fixing this relies on this PR being merged: https://github.com/Ultimaker/Cura/pull/2200

fieldOfView, 2017-08-14:
Resizing issue will be fixed with #53, which relies on Ultimaker/Cura#2200

closed 2017-08-14 by fieldOfView

netlands, 2017-09-05:
Verified on release version of Cura 2.7. The camera image now resizes when resizing the dialog.
Note that depending on the orientation (and aspect ratio?) of the image and the size of the dialog the resized image still overlaps with UI (for example a 90 degrees rotated image with a 16:9 aspect ratio).

fieldOfView, 2017-09-06:
Fixed in 536b5a780d33a59bb1661f71af53f6424c8066ec