fieldOfView/Cura-OctoPrintPlugin #32: Plugin fails to print from Cura 2.5

godefroi, 2017-04-13:
I'm using the 2.5 branch of the plugin with the Cura 2.5 beta, and while Cura will communicate with the printer and retrieve status just fine, it won't print. The progress bar appears momentarily, but OctoPrint never receives the file.

fieldOfView, 2017-04-13:
Are you using the 2.5 branch? You must use the branch that matches the Cura version

godefroi, 2017-04-13:
Yep, definitely. I updated it yesterday just to be sure.

fieldOfView, 2017-04-13:
Sorry for the cookie-cutter answer before. I'll look into it.
You haven't by chance enabled the SD-card option, have you?

godefroi, 2017-04-13:
I'm not sure what you mean by that. I don't have an SD card in my printer, and I've never printed from an SD before.

fieldOfView, 2017-04-13:
I was referring to a checkbox in the Connect OctoPrint interface.

godefroi, 2017-04-13:
Ah. No, the option is not selected. The "Automatically start..." box is selected, and the "Show webcam..." is not selected (though it seems to me that in previous 2.3 versions it must have defaulted to selected).

Cura correctly shows all the telemetry from OctoPrint (temperatures, status, job name, etc) on the printer tab.

godefroi, 2017-04-18:
More information: if I take an .STL on my hard drive, slice in Cura, and "Print to OctoPrint", it works great. If I model something in Fusion 360, "Save as STL" (which can send to Cura), then try to "Print to OctoPrint", it doesn't work.

fieldOfView, 2017-04-18:
Thanks, that is helpfull info.

I vaguely remember from an issue with Cura that when sending the print to Cura directly from Fusion 360, the filename comes in a something like {aec797d8-a1e8-45b5-8c47-4e9b949c7d95}.stl, which sets the job name in the lower right corner to {aec797d8-a1e8-45b5-8c47-4e9b949c7d95}, am I right? I guess the problem lies with the curly brackets. 

A workaround for now would be to remove them from the jobname in the lower right corner (you can edit that jobname just by clicking on it). If that fixes the issue for you, I will make sure the plugin sanitizes the jobname before sending it to OctoPrint.

fieldOfView, 2017-04-18:
Hmm, turns out there are no curly brackets. It might be the filename/jobname length then.

fieldOfView, 2017-04-18:
Anything helpful in the logs? I cannot reproduce this.

godefroi, 2017-04-18:
I'll get back to you today. Where could I look for logs?

fieldOfView, 2017-04-18:
In Cura, see Help -> Show Configuration Folder

There should be a file named Cura.log, which is probably fairly big. Please close Cura, remove that file, and then try printing through Fusion 360 -> Cura -> OctoPrint. Then close Cura, and post the log file, eg on pastebin.com

fieldOfView, 2017-04-18:
So it worked with 2.3, but not with 2.5? Did you try 2.4?

godefroi, 2017-04-19:
I never did try 2.4. Here's the cura.log: https://pastebin.com/CRj7RSqG

There's a stderr file as well, it seems to mostly have the same information.

godefroi, 2017-04-19:
Aha! Solid progress! If I enter a job name and *then* print, it works just fine.

fieldOfView, 2017-04-19:
Thanks for debugging. The issue is with Cura (which does not set a proper "jobname" when loading an object by specifying the filename in the command line), but I have added a workaround for the resulting issue.

Could you redownload the 2.5 version and try again?

fieldOfView, 2017-04-19:
See https://github.com/Ultimaker/Cura/issues/1703

godefroi, 2017-04-20:
Success. The job name is empty in Cura (as was your intent, I'm sure) and shows up in octoprint as untitled_print.gcode, and prints successfully. Thank you!

fieldOfView, 2017-04-20:
> The job name is empty in Cura (as was your intent, I'm sure)

The empty jobname is not something I can fix from the side of the OctoPrintPlugin, and will need to be fixed inside Cura. I have just worked around the issue here (and hope the workaround becomes superfluous over time). Thanks for reporting!

commit 6188c006 (2017-04-19): Make sure job name is not an empty string

This uses an untranslatable string, but it should really be fixed inside Cura. Fixes #32
commit 661d3cec (2017-04-19): Make sure job name is not an empty string

This uses an untranslatable string, but it should really be fixed inside Cura. Fixes #32
closed 2017-04-20 by fieldOfView