fieldOfView/Cura-OctoPrintPlugin #35: Pre-heat button fails on M80 printers

blole, 2017-04-29:
My printer requires the power to be turned on with `M80` before heating the bed. The fix could be as easy as just sending that command first, but I don't know if that might be something you don't want to do on printers that don't require it or if there are any other caveats.

fieldOfView, 2017-04-29:
I am not quite sure _where_ to fix this. Sure, I could add a checkmark in the Connect to OctoPrint screen to "turn on power before preheating", though that would be quite machine-specific, and it would not fix the same issue when you connect the printer directly via USB. It should be somehow implemented Cura-wide, for any connectiontype, instead of every connection-type ending up with a slew of fairly specific printer fixes.

Honestly I think it should really be fixed at the firmware level.

AlexAhrens, 2017-04-29:
Try putting the comment in the Start-Gcodes in Cura. Right in the first line.

fieldOfView, 2017-04-29:
Start G-Codes are not sent when using the preheat function.

blole, 2017-04-29:
Yes, you're right. I'm also thinking adding it in the firmware might be the best solution now.

fieldOfView, 2017-04-29:
Or perhaps there should be more gcode sequences in the Machine Settings window in Cura, such as Pause/Resume/Abort print, Preheat/End Preheat, etc, similar to how OctoPrint already has these.

In the mean time, you could experiment with the following:
https://github.com/fieldOfView/OctoPrintPlugin/blob/2.5/OctoPrintOutputDevice.py#L449
```
    def _setTargetBedTemperature(self, temperature):
        Logger.log("d", "Setting bed temperature to %s", temperature)
        if temperature > 0 : 
            self._sendCommand("M80")
        self._sendCommand("M140 S%s" % temperature)

    def _setTargetHotendTemperature(self, index, temperature):
        Logger.log("d", "Setting hotend %s temperature to %s", index, temperature)
        if temperature > 0 : 
            self._sendCommand("M80")
        self._sendCommand("M104 T%s S%s" % (index, temperature))
```

blole, 2017-04-29:
Yeah, that works :)

closed 2017-06-12 by fieldOfView

SirRedZ, 2019-05-08:
Which file do I edit on a windows installation of Cura 4.0 to achieve this result?