fieldOfView/Cura-OctoPrintPlugin #36: Plugin not working in new version Cura 2.5.0

averter, 2017-04-29:
Yesterday I did an update to the latest version of Cura 2.5.0, and this plugin stopped working.
I also reinstalled it via the command line, but nothing changed.

fieldOfView, 2017-04-29:
You don't specify if you downloaded the 2.5 version of the plugin. Changes were required, so you have to download the version of the plugin that matches your Cura version.

averter, 2017-04-29:
You are correct. I tried to update via PPA and effectively it was version 2.4, not 2.5. Is there a way to download the 2.5 version via PPA? Thanks.

fieldOfView, 2017-04-29:
pinging @thopiekar

thopiekar, 2017-04-29:
The reason why the plugin is not working is that I haven't had time to update the plugin package. The package itself probably says it is for version 2.5 but 2.4 is in there. There are more plugins where I have to check whether there is a 2.5 branch for it, but as I can see the next build of the octoprint plugin will come truely with 2.5. I'm surprised how many people are already using my PPA. Just found time before going to sleep to update it and next morning I had already an email and a Skype message :roll_eyes:

averter, 2017-04-29:
That doesn't surprises me. It's an excellent plugin :-)

thopiekar, 2017-04-29:
Well, I meant the full upgrade of my PPA to Cura 2.5...
However, the plugins are on the way: https://code.launchpad.net/~thopiekar/+recipe/cura-extra-plugins-stable-daily

averter, 2017-04-29:
Awesome! It is working already. This is what I call efficiency :)

closed 2017-04-29 by averter