fieldOfView/Cura-OctoPrintPlugin #37: Printing does not start

rpgrunwald, 2017-05-06:
Windows 10 64-Bit
Cura 2.5
Octoprint 1.3.2 on Raspian Jessie 
Printer Monoprice Select v2.1 (Prusa i3 selected)
Printer connects and I'm able to see temps.  Pre-heat button works.
"Print with Octoprint"  says it transfers but heaters do not turn on and job does not start. 
Cura can see a job started on Octoprint.  Camera feed shows in Cura. 
I did not see anything in grep -r octoprint in /var/log.  Where else should I look"
In cura.log all I see are 
2017-05-04 18:44:21,996 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 2.5\resources\shaders\transparent_object.shader]...
2017-05-04 18:44:23,715 - DEBUG - CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [447]: Slicing took 2.537142276763916 seconds
2017-05-04 18:44:23,720 - DEBUG - UM.Backend.Backend._logSocketState [176]: Socket state changed to Closing
2017-05-04 18:44:23,723 - DEBUG - UM.Backend.Backend._onSocketError [198]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2017-05-04 18:44:23,727 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Closed
2017-05-04 18:44:23,729 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [545]: Backend quit with return code 0. Resetting process and socket.
2017-05-04 18:44:23,909 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:25,919 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:27,894 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:28,800 - DEBUG - LocalFileOutputDevice.LocalFileOutputDevicePlugin.requestWrite [126]: Writing to [C:/Users/Paul/Desktop/3D Printing/pi base.gcode]...
2017-05-04 18:44:29,914 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:30,468 - DEBUG - LocalFileOutputDevice.LocalFileOutputDevicePlugin.requestWrite [152]: Writing to Local File C:/Users/Paul/Desktop/3D Printing/pi base.gcode in text mode
2017-05-04 18:44:30,468 - DEBUG - SliceInfoPlugin.SliceInfo.run [51]: Sending anonymous slice info to [https://stats.youmagine.com/curastats/slice]...
2017-05-04 18:44:30,708 - DEBUG - UM.FileHandler.WriteFileJob.run [41]: Writing file took 0.2130270004272461 seconds
2017-05-04 18:44:31,889 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:32,546 - INFO - SliceInfoPlugin.SliceInfo.run [55]: Sent anonymous slice info.
2017-05-04 18:44:33,900 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:35,895 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:37,886 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:38,802 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-05-04 18:44:39,912 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:41,925 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:43,902 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:45,926 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:46,553 - DEBUG - LocalFileOutputDevice.LocalFileOutputDevicePlugin.requestWrite [126]: Writing to [C:/Users/Paul/Desktop/3D Printing/pi base.gcode]...
2017-05-04 18:44:47,912 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:48,697 - DEBUG - LocalFileOutputDevice.LocalFileOutputDevicePlugin.requestWrite [155]: Writing to Local File C:/Users/Paul/Desktop/3D Printing/pi base.gcode in binary mode
2017-05-04 18:44:48,698 - DEBUG - SliceInfoPlugin.SliceInfo.run [51]: Sending anonymous slice info to [https://stats.youmagine.com/curastats/slice]...
2017-05-04 18:44:49,098 - DEBUG - UM.FileHandler.WriteFileJob.run [41]: Writing file took 0.375058650970459 seconds
2017-05-04 18:44:49,920 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:50,832 - INFO - SliceInfoPlugin.SliceInfo.run [55]: Sent anonymous slice info.
2017-05-04 18:44:51,892 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:53,947 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502
2017-05-04 18:44:55,937 - WARNING - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [553]: Received an unexpected returncode: 502

What else should I look at?

fieldOfView, 2017-05-06:
Those 502 returncodes are worrysome.

>  502 Bad Gateway
The server, while acting as a gateway or proxy, received an invalid response from the upstream server it accessed in attempting to fulfill the request.

Is OctoPrint running behind a proxy?
Can you see the file created in OctoPrint?

rpgrunwald, 2017-05-06:
I'm not behind a proxy.  This is on a lan. 
Where would the file be stored by default by Octoprint?

fieldOfView, 2017-05-06:
The gcode file should end up in the file list, as if you manually uploaded it. It should also end up in your filesystem, in ~/.octoprint/uploads.

What port is OctoPrint running on?

rpgrunwald, 2017-05-06:
I do not see the file in the file list in Octopi.   ~/.octoprint/uploads only has manually uploaded files. 

I used the stock installation http://octoprint.org/download/ - the only thing I have changed is a "moderate" overclock and apt-get update and apt-get dist-upgrade.  Octoprint updated to 1.3.2 through the GUI.   Some plugins listed here were installed. http://3dprinterwiki.info/wiki/wanhao-duplicator-i3/computer-software/octoprint-on-raspberry-pi/ 
I do have a new user created and using enhanced security.  I used the new API key in Cura. 

I did a fresh install on a fully formated SD card last night so this is "clean"

`pi@octopi:/etc/haproxy $ more haproxy.cfg
global
        maxconn 4096
        user haproxy
        group haproxy
        log 127.0.0.1 local1 debug

defaults
        log     global
        mode    http
        option  httplog
        option  dontlognull
        retries 3
        option redispatch
        option http-server-close
        option forwardfor
        maxconn 2000
        timeout connect 5s
        timeout client  15min
        timeout server  15min

frontend public
        bind *:80
        bind 0.0.0.0:443 ssl crt /etc/ssl/snakeoil.pem
        option forwardfor except 127.0.0.1
        use_backend webcam if { path_beg /webcam/ }
        default_backend octoprint
        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

backend octoprint
        reqrep ^([^\ :]*)\ /(.*) \1\ /\2
        reqadd X-Scheme:\ https if { ssl_fc }
        option forwardfor
        server octoprint1 127.0.0.1:5000

backend webcam
        reqrep ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8080
`
~/.octoprint/uploads only has manually uploaded file. 

cura.log:

2017-05-06 11:03:04,588 - INFO - UM.Backend.Backend.startEngine [72]: Started engine process: C:\Program Files\Cura 2.5\CuraEngine.exe
2017-05-06 11:03:04,593 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] Calling engine with: ['C:\\Program Files\\Cura 2.5\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Cura 2.5\\resources\\definitions\\fdmprinter.def.json', '']
2017-05-06 11:03:04,646 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] 
2017-05-06 11:03:04,650 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] Cura_SteamEngine version DEV
2017-05-06 11:03:04,655 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] Copyright (C) 2014 David Braam
2017-05-06 11:03:04,662 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] 
2017-05-06 11:03:04,685 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] This program is free software: you can redistribute it and/or modify
2017-05-06 11:03:04,689 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-05-06 11:03:04,693 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-05-06 11:03:04,697 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] (at your option) any later version.
2017-05-06 11:03:04,700 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] 
2017-05-06 11:03:04,704 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] This program is distributed in the hope that it will be useful,
2017-05-06 11:03:04,708 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2017-05-06 11:03:04,711 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2017-05-06 11:03:04,715 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] GNU Affero General Public License for more details.
2017-05-06 11:03:04,719 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] 
2017-05-06 11:03:04,724 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] You should have received a copy of the GNU Affero General Public License
2017-05-06 11:03:04,728 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-05-06 11:03:04,760 - DEBUG - UM.Backend.Backend._logSocketState [172]: Socket state changed to Connected
2017-05-06 11:03:04,763 - DEBUG - UM.Backend.Backend._onSocketStateChanged [162]: Backend connected on port 49674
[... 24 lines cut ...]
2017-05-06 11:03:07,787 - DEBUG - UM.Backend.Backend._onSocketError [198]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2017-05-06 11:03:07,790 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Closed
2017-05-06 11:03:07,801 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [545]: Backend quit with return code 0. Resetting process and socket.
2017-05-06 11:03:14,535 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-05-06 11:03:27,957 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-05-06 11:03:32,935 - DEBUG - OctoPrintPlugin.OctoPrintOutputDevicePlugin._onServiceChanged [152]: Bonjour service added: octopi._octoprint._tcp.local
2017-05-06 11:03:32,966 - WARNING - cura.PrinterOutputDevice.address [176]: address is not implemented by this output device.
2017-05-06 11:03:32,969 - WARNING - cura.PrinterOutputDevice.address [176]: address is not implemented by this output device.
[... cut ...]
2017-05-06 11:04:53,691 - WARNING - cura.PrinterOutputDevice.address [176]: address is not implemented by this output device.
2017-05-06 11:04:53,695 - WARNING - cura.PrinterOutputDevice.address [176]: address is not implemented by this output device.
2017-05-06 11:04:53,699 - WARNING - cura.PrinterOutputDevice.address [176]: address is not implemented by this output device.
2017-05-06 11:04:55,428 - DEBUG - UM.Backend.Backend._createSocket [209]: Previous socket existed. Closing that first.
2017-05-06 11:04:55,503 - DEBUG - UM.Backend.Backend._logSocketState [168]: Socket state changed to Listening

fieldOfView, 2017-05-07:
Well, at least it is no longer throwing up these 502 errors.

Something else to look into are the logs in ~/.octoprint/logs

rpgrunwald, 2017-05-07:
octoprint.log
[octoprint (1).zip](https://github.com/fieldOfView/OctoPrintPlugin/files/982032/octoprint.1.zip)

fieldOfView, 2017-05-08:
The logs have a number of 409 errors in them, when it could not connect to the printer (ie: connection between the printer and your pi did not work).

I am somewhat out of ideas. The logs show you have a number of plugins installed. Can you try without the plugins?

You don't have the "Store gcode on SD-card" option checked, do you?

fieldOfView, 2017-05-08:
> You don't have the "Store gcode on SD-card" option checked, do you?

The logs suggest you do have that option checked. This refers to the SD-Card of your printer, not the rpi SD-Card. Transferring the gcode file from the Pi to the printer takes *very* long, almost as long as printing the same gcode. I should probably add a warning to that option (or remove it altogether).

Final question: is does the monitor tab/button in the upper right corner have a green or a gray overlay circle?

rpgrunwald, 2017-05-09:
I did have the store on G-code selected but no card installed.   That was the problem.  I was always getting green.   It is now working.   I'm sorry for the trouble,  I assumed that it would ignore that setting because it said: "if available".  I didn't even think to uncheck it.  

Anyway, this is very cool!  Thank you for your work!  I appreciate it!

closed 2017-05-09 by fieldOfView

xkisu, 2017-06-07:
I'm having a similar issue with it uploading to the SD card. Where can I find the "Store gcode on SD-card" option?

rpgrunwald, 2017-06-07:
I'm not in front of my computer at home but I think it is in Cura's 
"manage printers" where you put the Octoprint IP address.

xkisu, 2017-06-07:
Yeah it is, thanks. I found it a few minutes after posting :)