fieldOfView/Cura-OctoPrintPlugin #39: Getting lot of KeyError: "temperature"

rafaelrs, 2017-06-06:
`2017-06-07 01:29:03,271 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [134]: Creating additional ui components for UM3.
2017-06-07 01:29:03,297 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction._createAdditionalComponentsView [142]: Creating additional ui components for OctoPrint-connected printers.
2017-06-07 01:29:03,725 - DEBUG - UM.Qt.QtRenderer._initialize [172]: Support for Vertex Array Objects: True
2017-06-07 01:29:03,731 - DEBUG - UM.View.GL.OpenGL.__init__ [77]: Initialized OpenGL subsystems.
2017-06-07 01:29:03,734 - DEBUG - UM.View.GL.OpenGL.__init__ [78]: OpenGL Version:  4.5 (Core Profile) Mesa 17.0.5
2017-06-07 01:29:03,735 - DEBUG - UM.View.GL.OpenGL.__init__ [79]: OpenGL Vendor:   Intel Open Source Technology Center
2017-06-07 01:29:03,736 - DEBUG - UM.View.GL.OpenGL.__init__ [80]: OpenGL Renderer: Mesa DRI Intel(R) HD Graphics 530 (Skylake GT2) 
2017-06-07 01:29:03,738 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/resources/shaders/default.shader]...
2017-06-07 01:29:03,746 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/resources/shaders/selection.shader]...
2017-06-07 01:29:03,751 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/resources/shaders/default.shader]...
2017-06-07 01:29:03,758 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/resources/shaders/composite.shader]...
2017-06-07 01:29:03,775 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/resources/shaders/overhang.shader]...
2017-06-07 01:29:03,795 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/resources/shaders/striped.shader]...
2017-06-07 01:29:03,817 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/resources/shaders/default.shader]...
2017-06-07 01:29:03,824 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/resources/shaders/grid.shader]...
2017-06-07 01:29:03,832 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/resources/shaders/platform.shader]...
2017-06-07 01:29:04,018 - WARNING - CuraEngineBackend.StartSliceJob.run [129]: No objects suitable for one at a time found, or no correct order found
2017-06-07 01:29:04,221 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-06-07 01:29:04,222 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-06-07 01:29:04,223 - CRITICAL - cura.CrashHandler.show [39]:   File "/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/plugins/plugins/OctoPrintPlugin/OctoPrintOutputDevice.py", line 527, in _onRequestFinished
2017-06-07 01:29:04,224 - CRITICAL - cura.CrashHandler.show [39]:     temperature = json_data["temperature"]["tool%d" % index]["actual"] if ("tool%d" % index) in json_data["temperature"] else 0
2017-06-07 01:29:04,226 - CRITICAL - cura.CrashHandler.show [39]: KeyError: 'temperature'
2017-06-07 01:29:06,376 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-06-07 01:29:06,377 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-06-07 01:29:06,379 - CRITICAL - cura.CrashHandler.show [39]:   File "/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/plugins/plugins/OctoPrintPlugin/OctoPrintOutputDevice.py", line 527, in _onRequestFinished
2017-06-07 01:29:06,380 - CRITICAL - cura.CrashHandler.show [39]:     temperature = json_data["temperature"]["tool%d" % index]["actual"] if ("tool%d" % index) in json_data["temperature"] else 0
2017-06-07 01:29:06,382 - CRITICAL - cura.CrashHandler.show [39]: KeyError: 'temperature'
2017-06-07 01:29:08,165 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-06-07 01:29:08,166 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-06-07 01:29:08,167 - CRITICAL - cura.CrashHandler.show [39]:   File "/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/plugins/plugins/OctoPrintPlugin/OctoPrintOutputDevice.py", line 527, in _onRequestFinished
2017-06-07 01:29:08,168 - CRITICAL - cura.CrashHandler.show [39]:     temperature = json_data["temperature"]["tool%d" % index]["actual"] if ("tool%d" % index) in json_data["temperature"] else 0
2017-06-07 01:29:08,169 - CRITICAL - cura.CrashHandler.show [39]: KeyError: 'temperature'
2017-06-07 01:29:10,184 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-06-07 01:29:10,186 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-06-07 01:29:10,187 - CRITICAL - cura.CrashHandler.show [39]:   File "/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/plugins/plugins/OctoPrintPlugin/OctoPrintOutputDevice.py", line 527, in _onRequestFinished
2017-06-07 01:29:10,188 - CRITICAL - cura.CrashHandler.show [39]:     temperature = json_data["temperature"]["tool%d" % index]["actual"] if ("tool%d" % index) in json_data["temperature"] else 0
2017-06-07 01:29:10,189 - CRITICAL - cura.CrashHandler.show [39]: KeyError: 'temperature'
2017-06-07 01:29:12,167 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-06-07 01:29:12,168 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-06-07 01:29:12,170 - CRITICAL - cura.CrashHandler.show [39]:   File "/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/plugins/plugins/OctoPrintPlugin/OctoPrintOutputDevice.py", line 527, in _onRequestFinished
2017-06-07 01:29:12,171 - CRITICAL - cura.CrashHandler.show [39]:     temperature = json_data["temperature"]["tool%d" % index]["actual"] if ("tool%d" % index) in json_data["temperature"] else 0
2017-06-07 01:29:12,173 - CRITICAL - cura.CrashHandler.show [39]: KeyError: 'temperature'
`

Linux - Fedora 25
OctoPrint 1.3.4
Cura 2.5.0 (here i didn't understood how to put plugin to app when i start it from appImage, i just unpacked this appImage, putted plugin to ./Cura-2.5.0/usr/bin/plugins/plugins/OctoPrintPlugin and started ./Cura-2.5.0/AppRun )

fieldOfView, 2017-06-10:
I'll look in to it. Apparently your octoprint instance is reporting data, but no temperatures. Anything special about the printer I should know about? Is the plugin working otherwise? (ie: can you send a print to octoprint from inside Cura?)

On installing the plugin, you can put it in the configuration folder (~/.local/shared/cura/plugins ?) 
That folder should alreadt have a subfolder named "plugins".

fieldOfView, 2017-06-10:
The errors should be fixed now, in the branches 2.5, 2.6 and master.

rafaelrs, 2017-06-10:
Yes, i can send to print and it prints good. But it was showing no printing progress, and seems like not always catches temperatures (i didn't saw hotend temp ever for example). Nothing special in my printer, it's just FLSUN cube printer based on MKS Gen-L board with Marling 1.1.0 RC7

Just tried your last fix, still don't see hotend temp, and printing progress is zero always and i see these errors:

`2017-06-11 00:56:57,997 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-06-11 00:56:57,998 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-06-11 00:56:58,000 - CRITICAL - cura.CrashHandler.show [39]:   File "/home/rafaelrs/AddProgs/Cura/Cura-2.5.0/usr/bin/plugins/plugins/OctoPrintPlugin/OctoPrintOutputDevice.py", line 535, in _onRequestFinished
2017-06-11 00:56:58,001 - CRITICAL - cura.CrashHandler.show [39]:     if json_data["state"]["flags"]["error"]:
2017-06-11 00:56:58,002 - CRITICAL - cura.CrashHandler.show [39]: TypeError: string indices must be integers
`

That's how it looks: [http://storage2.static.itmages.com/i/17/0610/h_1497132490_9360510_a7f0d01ff7.png](url)

fieldOfView, 2017-06-11:
Could you try the following URL in your browser, and send me the results?
```http://[octoprint address]/api/printer?apikey=[api-key]```
eg: ```http://octopi.local/api/printer?apikey=73C7E2006D25AA71AE31239F28549F14```

rafaelrs, 2017-06-11:
Sorry for delay, just reached my printer output:

`$ curl http://3dprinter.home:5000/api/printer?apikey=********************************
{
  "sd": {
    "ready": false
  }, 
  "state": {
    "flags": {
      "closedOrError": false, 
      "error": false, 
      "operational": true, 
      "paused": false, 
      "printing": true, 
      "ready": false, 
      "sdReady": false
    }, 
    "text": "Printing"
  }, 
  "temperature": {
    "bed": {
      "actual": 45.4, 
      "offset": 0, 
      "target": 45.0
    }, 
    "tool0": {
      "actual": 214.9, 
      "offset": 0, 
      "target": 215.0
    }
  }
}`

fieldOfView, 2017-06-12:
That looks like correct output, but apparently it is not what the plugin is getting from OctoPrint. I'll add some more logging to find out what is going on.

fieldOfView, 2017-06-12:
I have updated the 2.5 branch to add some more logging. Could you try it out? I am specifically looking for debug log entries that start with ```Received incomplete json from OctoPrint:```

rafaelrs, 2017-06-12:
Here you go:

`2017-06-12 22:54:29,611 - DEBUG - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [551]: Received incomplete json from OctoPrint: {'progress': {'filepos': None, 'completion': None, 'printTime': None, 'printTimeLeft': None}, 'state': 'Printing', 'job': {'estimatedPrintTime': None, 'lastPrintTime': None, 'filament': None, 'averagePrintTime': None, 'file': {'origin': 'local', 'path': 'FLP_auffahrt.gcode', 'date': 1497297242, 'name': 'FLP_auffahrt.gcode', 'size': 1702449}}}
2017-06-12 22:54:31,558 - DEBUG - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [551]: Received incomplete json from OctoPrint: {'progress': {'filepos': None, 'completion': None, 'printTime': None, 'printTimeLeft': None}, 'state': 'Printing', 'job': {'estimatedPrintTime': None, 'lastPrintTime': None, 'filament': None, 'averagePrintTime': None, 'file': {'origin': 'local', 'path': 'FLP_auffahrt.gcode', 'date': 1497297242, 'name': 'FLP_auffahrt.gcode', 'size': 1702449}}}
`

fieldOfView, 2017-06-12:
Doh!

It breaks because you have the string "printer" in the hostname. I'll create a fix.

fieldOfView, 2017-06-12:
Should be fixed now. Can you give it another go?

rafaelrs, 2017-06-12:
Just checked. Works fine, shows everything. Thanks a lot!

fieldOfView, 2017-06-12:
Thanks for helping me get to the bottom of this.

closed 2017-06-12 by fieldOfView