fieldOfView/Cura-OctoPrintPlugin #40: No video in octoprint preview

fieldOfView, 2017-06-10:
As reported by @MaukCC here: https://github.com/Ultimaker/Cura/issues/1947

OS ubuntu 16.04
Cura 2.5.99 master

Since a couple of days the video feed from Octoprint is gone in Cura.
Also when setting Octoprint up, sometimes you can see the xxxx_octoprint._tcp.local but most of the times it is not there and you manually have to enter the IP address.

fieldOfView, 2017-06-10:
See this commit: https://github.com/Ultimaker/Cura/commit/ca922e9e9da6a9209d8a27849f7ca1949ff9379a

fieldOfView, 2017-06-10:
Should be fixed here: https://github.com/fieldOfView/OctoPrintPlugin/commit/48a5819a031343433212d9e07d22e8404322ad6b

Can't test currently because I am away from my octoprint instance and it currently has no camera attached. @maukcc, can you test?

maukcc, 2017-06-12:
Yes that worked. Video is back.
The changes to OctoPrintOutputDevice.py where already done with the latest update.
The new MonitorItem.qml file was not added with the latest update.

fieldOfView, 2017-06-12:
> The new MonitorItem.qml file was not added with the latest update.

Thanks for mentioning that; I forgot to add it to the cmake config

closed 2017-06-12 by fieldOfView