fieldOfView/Cura-OctoPrintPlugin #41: files are not being send to octoprint

maukcc, 2017-06-15:
Everything works, except it is not sending the file to the octoprint server

fieldOfView, 2017-06-15:
What version of Cura, what version of the OctoPrintPlugin?
Do the logs say anything useful?

maukcc, 2017-06-15:
2.5.99 on ubuntu 16.04 updated yesterday
OctoPrintPlugin 1.3.4

log:
2017-06-15 12:18:10,732 - DEBUG - SliceInfoPlugin.SliceInfo.run [51]: Sending anonymous slice info to [https://stats.youmagine.com/curastats/slice]...
2017-06-15 12:18:10,737 - WARNING - cura.PrinterOutputDevice.address [212]: address is not implemented by this output device.
2017-06-15 12:18:10,738 - WARNING - cura.PrinterOutputDevice.address [212]: address is not implemented by this output device.
2017-06-15 12:18:10,743 - ERROR - OctoPrintPlugin.OctoPrintOutputDevice.startPrint [439]: An exception occurred in network connection: name 'catalog' is not defined
2017-06-15 12:18:11,906 - ERROR - SliceInfoPlugin.SliceInfo.run [58]: An HTTP error occurred while trying to send slice information: HTTP Error 500: Internal Server Error

fieldOfView, 2017-06-15:
Whoops, I hope I fixed that in time for the final 2.6 build. Thanks for reporting.

fieldOfView, 2017-06-15:
Fixed here: fc9cd78905af84b60c7c5f0a7cf5c9470253017e

closed 2017-06-15 by fieldOfView

averter, 2017-08-12:
Sorry to revisit this issue. For some reason I am affected by this same bug. 
I was trying to use the PPA of OctoprintPlugin to update to version 2.6 and get it right, but it seems it's still on 2.4.
Alternatively I have a manual install which is on 2.6 but I don't know how to swap between the version installed by the PPA and the manual one. Can someone help me please?

fieldOfView, 2017-08-12:
I know nothing about PPAs, so I am going to ping @thopiekar for that.

In the mean time, try using the AppImage. The PPA is not officially supported.

thopiekar, 2017-08-15:
Using the PPA should make no difference. Like in the AppImage, all Cura* files are readonly.

fieldOfView, 2017-08-15:
@averter seems to suggest that his 2.6 PPA installation "comes with" the OctoPrintPlugin for Cura 2.4 (ie: the 2.4 branch instead of the 2.6 branch).

thopiekar, 2017-08-15:
@fieldOfView Shouldn't be the case: https://code.launchpad.net/~thopiekar/+recipe/cura-extra-plugins-stable-daily

averter, 2017-08-15:
Oh great. I've done some silly mistake while trying to update via PPA  and I ended up scrambling cura. I cannot even start it.

averter, 2017-08-15:
Nope, it's still on 2.4. How can I get the version that you just shown? Shouldn't an apt-get update do the trick of updating the PPA?

thopiekar, 2017-08-15:
Well, what I can imagine of is that the upgrade of the plugin is blocked. Take a GUI like Synaptic and check whether the package is really updated or not.

averter, 2017-08-15:
Thanks :-). I think it is working now. There were two plugins: "Cura-extra-plugin-octoprint" the 2.4 one, and the "Cura-plugin-octoprint" the 2.6 version. I've switched from one to the other and I'm going to test it when at home. I will let you know.

thopiekar, 2017-08-15:
It is a long time ago, since I renamed the plugin packages, but, however,
glad to hear it worked.

maukcc, 2017-08-15:
Is there any reason why these extra plugins are not in the Cura ppa? (they are in the appimage)
Then we would not have these uncertainties about what plugin belongs to what version of Cura

thopiekar, 2017-08-15:
I just renamed the packages. It can sometimes happen that one of these
extra packages can get included into Cura itself - at least in the past it
was more likely.
To avoid any further conflicts I just decided to call all plugin packages
in a common way.

fieldOfView, 2017-08-15:
Note that there's an issuequeue for the PPA packages here:
https://github.com/thopiekar/Cura-packaging/issues
The PPA (and the official AppImage for that matter) are out of my control.

averter, 2017-08-15:
I strongly suggest to update the README file on the "how to install" section, so that you install "Cura-plugin-octoprint", and not the extra outdated version. It could be misleading.

fieldOfView, 2017-08-15:
@thopiekar could you make a pull request to correct the installation instructions? Thanks.