fieldOfView/Cura-OctoPrintPlugin #43: webcam is not shown in Cura if the ip address is different to octaprint

andrasbiro, 2017-06-25:
As per the title. I'm not 100% sure the issue is with the plugin, but I did see the webcam picture when the webcam server was the same as the octaprint server. Now that the the camera has a different ip, I don't see the webcam.
(my setup has an old raspberry pi, which has the camera and forwards the serial port to TCP port, and octaprint runs on my home server, connecting to both the camera and the TCP server on the raspi)

fieldOfView, 2017-06-25:
This has been fixed for the version of the plugin that will ship with Cura 2.7

closed 2017-06-25 by fieldOfView

fieldOfView, 2017-06-25:
OctoPrintPlugin 2.7 gets the webcam-configuration from OctoPrint, including server, port and image orientation. The version for Cura 2.6 just uses a non-configurable default url.

andrasbiro, 2017-06-25:
Understood, I will wait for 2.7