fieldOfView/Cura-OctoPrintPlugin #45: Add support for HTTP-Basic authentification

jcastang, 2017-07-10:
Hello,

Can you please add option to allow http-basic Auth ?
Login and password can be passed through url but textbox do not allow special characters lis "@".

Thanks.

fieldOfView, 2017-07-10:
I have put this on my list for the release that will ship with Cura 2.7

For now, you could try editing cura.cfg.

jcastang, 2017-07-10:
Just to say: editing cura.cfg works. 
Thanks for the tip.

fieldOfView, 2017-07-10:
Note to self: https://stackoverflow.com/a/1700751

fieldOfView, 2017-07-14:
I have implemented this, but it needs testing. Unfortunately, testing can only be done with Cura 2.7, which is not yet available.

fieldOfView, 2017-07-24:
Left-over issue: add basic authentication support to the API key check in discovery.

fieldOfView, 2017-08-02:
A beta of Cura 2.7 is now available. I would appreciate if you could test this functionality

AidanCodes, 2017-09-07:
Is the API key check still having auth issues? I am having issues getting it to work and I'm curious as to is it was ever fixed

fieldOfView, 2017-09-07:
Basic Auth is implemented, but not tested much. I have not configured my octoprint installation with authentication. That's why I asked for testing above.

If you have a guide/link that shows me how to configure haproxy for basic auth, I could see what is going on.

fieldOfView, 2017-09-07:
Basic Authentication turns out to be broken in the 2.7 release. I have fixed it here: https://github.com/fieldOfView/OctoPrintPlugin/commit/b26437b0c32dcb7ee4b831e5ea1bb40afc672885

AidanCodes, 2017-09-07:
Thank you so much! Your quick response (and great work and contributions to the 3D printing community) are highly appreciated! That being said, in testing I still came across some feature issues when using basic auth including webcam functionality. Is there a better place to put such comments? (I apologize, I am fairly new to finding/testing bugs and submitting reports)

fieldOfView, 2017-09-07:
Here will do. I am reopening this issue, so we can get all issues out of Basic Auth

fieldOfView, 2017-09-07:
Tell me about your setup. What is your octoprint URL, and what is your webcam URL? Do they share the same Basic Auth?

AidanCodes, 2017-09-07:
Thanks again. I'm running octoprint through haproxy to port 80 on http://<localPrinterUrl>/. Webcam runs to http://<localPrinterUrl>/webcam/?action=stream. They share the same auth.

[haproxy.cfg gist](https://gist.github.com/AidanCodes/009998d44f15228bdc1603ef010196b9)

fieldOfView, 2017-09-07:
I sharing the same basic auth between octoprint and the webcam works for me, with the same haproxy ports and adresses. I have a slightly different haproxy setup, derived from this: https://www.ohbyteme.com/posts/using-haproxy-http-basic-authentication-to-secure-access-to-kibana

https://pastebin.com/zBTcnrGC

closed 2017-11-11 by fieldOfView