fieldOfView/Cura-OctoPrintPlugin #46: Handle multiple instances with different API keys more intuitively

fieldOfView, 2017-07-14:
The UI currently does not store the API key with the OctoPrint instance, but with the printer instance. It makes more sense to store the API key with the OctoPrint instance, because then it is easier to show which key was entered for which instance.

CPS-3, 2018-04-28:
When can we expect this issue to be resolved, because i see this issue being removed and added from Cura version to Cura version. Glad someone is known to this issue and will adress it...

CPS-3, 2018-05-18:
How do you set it to be the same. Can't really figure it out.

jacycatlin, 2018-06-18:
Yeah, the field is greyed out - I tried forcing it to accept input via Chrome's Inspect feature, but when I enter the new key it doesn't save.

fieldOfView, 2018-06-18:
Ok, that used to work. I guess now the only way is to edit the OctoPrint configuration file.

I'll put this issue back on top of my list.

jacycatlin, 2018-06-18:
Hey, editing the .yaml file and rebooting seems to have done the trick!  For now, anyway.  Thanks for helping to shave about 4 steps off my workflow - and here's hoping the final fix arrives someday to get rid of 2-3 more...

AkiDK, 2018-08-07:
did the change Both printers print at the same time :(

fieldOfView, 2018-08-21:
This will be fixed in the release for the next version of Cura.

commit 683cc937 (2018-08-21): Keep a cache of valid keys so facilitate switching

Fixes #46