fieldOfView/Cura-OctoPrintPlugin #48: enable user access control printing

gurusonwheels, 2017-07-17:
i have just downloaded 2.6 of cura . 
however the new octoprint does not seem to work .
seems that this new feature does not work on octoprint installs with access control enabled .
i get this error in cura : octoprint on octopi._octoprint._tcp.local does not allow access to print.

fieldOfView, 2017-07-17:
Did you enter a complete API key?

The OctoPrintPlugin needs an API key to communicate with OctoPrint. This can either be the system API key (as shown in Settings -> API) or the key of the user (as shown in User Settings or Settings -> Access Control - the latter is the easiest to copy/paste)

fieldOfView, 2017-07-17:
The version of OctoPrintPlugin that will ship with Cura 2.7 checks the API key after entering it, and does not let you continue if the API key is not accepted.

fieldOfView, 2017-07-18:
Please let me know if this fixed your issue.

closed 2017-07-19 by fieldOfView