fieldOfView/Cura-OctoPrintPlugin #49: Send user agent string

probonopd, 2017-07-23:
Please send a more informative user agent string.

```
POST /api/files/local HTTP/1.1
(...)
User-Agent: Mozilla/5.0
(...)
```

fieldOfView, 2017-07-23:
Can you describe why you want/need this?

probonopd, 2017-07-23:
I am implementing a compatible server speaking the OctoPrint protocol, and to be able to determine which kind of client is resulting in which bugs that would be very useful for me.

fieldOfView, 2017-07-23:
Any suggestion what the User Agent should be/include?

probonopd, 2017-07-23:
`User-Agent: OctoPrintPlugin/<VERSION>` maybe?

nallath, 2017-07-24:
Cura uses Cura-<VERSION>, so OctoPrintPlugin-<VERSION> would be my preference, but only slightly so.

fieldOfView, 2017-07-24:
Looking at most common user agents, ```Cura/2.7``` would be more in line than ```Cura - 2.7```.
I propose ```Cura/2.7 OctoPrintPlugin/2.7``` for this plugin.

fieldOfView, 2017-07-24:
Implemented in the master branch, which will roll into the 2.7 branch shortly.

closed 2017-07-24 by probonopd