fieldOfView/Cura-OctoPrintPlugin #52: Switching off the printer in an emergency crashes Cura

probonopd, 2017-08-13:
Switching off the printer in an emergency crashes Cura. It would be good if all functions accessing the OctoPrint API would be wrapped in `try:` statements.

```
Version: 2.7.0-BETA
Platform: Linux-4.4.0-31-generic-x86_64-with-debian-stretch-sid
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/tmp/.mount_Cura_fpsEW8j/usr/bin/plugins/plugins/OctoPrintPlugin/OctoPrintOutputDevice.py", line 569, in _onRequestFinished
    json_data = json.loads(bytes(reply.readAll()).decode("utf-8"))
  File "/build/env/2.7/inst/lib/python3.5/json/__init__.py", line 319, in loads
  File "/build/env/2.7/inst/lib/python3.5/json/decoder.py", line 339, in decode
  File "/build/env/2.7/inst/lib/python3.5/json/decoder.py", line 355, in raw_decode
json.decoder.JSONDecodeError: Expecting ',' delimiter: line 4 column 19 (char 57)

```

fieldOfView, 2017-08-14:
Fixed in 38e5a31

closed 2017-08-14 by fieldOfView

fieldOfView, 2017-08-14:
Note that with an actual OctoPrint instance, turning off the printer does not crash Cura. Turning off the computer that is running OctoPrint might though.

probonopd, 2017-08-14:
Thank you very much.

> Note that with an actual OctoPrint instance, turning off the printer does not crash Cura. Turning off the computer that is running OctoPrint might though.

That's what I meant. I am running mine from the printer's power supply.

commit 38e5a311 (2017-08-14): Make plugin robust against invalid JSON data returned by the instance

Fixes #52