fieldOfView/Cura-OctoPrintPlugin #57: When using first layer bed temp, plugin displays "normal" bed target temp

spinkham, 2017-08-28:
When preheating and printing first layer, plugin displays regular build plate temp as target rather than first layer temp target printer is actually using at that time.

Version tested: Cura daily PPA, master (2.6.99-master\~201708250033\~rev2169\~pkg163\~ubuntu17.04.1)
cura-plugin-octoprint (2.6.99\~201708161146\~159\~ubuntu17.04.1)

fieldOfView, 2017-08-28:
Could you check if the temperature is correctly shown in the OctoPrint web interface?

spinkham, 2017-08-28:
OctoPrint web "target" temp is correct.

fieldOfView, 2017-08-28:
Turns out the plugin is not looking at the target temperature at all. 
I'll fix this for the 2.8 release.

fieldOfView, 2017-09-01:
This should be fixed for the next release. The next release of Cura will also show target temperatures for the hotend(s).

closed 2017-09-01 by fieldOfView