fieldOfView/Cura-OctoPrintPlugin #58: When first connecting to OctoPrint, Cura will crash if the Stream url is blank.

JoelBrenstrum, 2017-09-11:
Looks to be line 665 in OctoPrintOutputDevice.py
elif stream_url[:4].lower() == "http":

fieldOfView, 2017-09-11:
That is strange, because the if statement above that line checks for the stream_url being empty or None.
Could you post the contents of http://[octoprint url]/api/settings?apikey=[your key]?
(you may want to edit out the api key that is included in it)

JoelBrenstrum, 2017-09-11:
Yea I noticed that too. after adding a url and removing it the issue doesnt happen (im unable to get he logs). This was a fresh OctoPrint install, I wonder if the settings object is the actual cause of the issue
`stream_url = json_data["webcam"]["streamUrl"]` 
Maybe the webcam object doesn't exist until a property has been set.

fieldOfView, 2017-09-11:
Can you post your logs (eg on pastebin.com)? Otherwise we'll remain guessing.

fieldOfView, 2017-09-27:
If it can't be reproduced, and logs cannot be found, I cannot fix it.

closed 2017-09-27 by fieldOfView