fieldOfView/Cura-OctoPrintPlugin #60: Cura settings not included in Gcode file when printing to OctoPrint

arose62, 2017-09-27:
Ubuntu 16.04 LTS, Cura 2.7.0

If I print from Cura to OctoPrint, the file does not include the Cura settings strings after the "Ending Gcode", but if I save the file, then send it to OctoPrint, the settings ARE included.

I think they should ALWAYS be included.

For example: the last lines below, which start with ";SETTING_3" are NOT sent directly to OctoPrint, but were included when I saved to file.

G0 F7200 X106.419 Y62.948
G0 X106.461 Y62.943
G0 X106.525 Y62.959
;TIME_ELAPSED:5158.739727
G1 F1800 E2961.34532
M107
;End Gcode ALR 27-SEP-2017
M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+15 E-2 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
M107 ; fan off
G90 ;absolute positioning
G1 Y190 ;present the finished print job
M84 ;steppers off
M104 S0
;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 2\\nname = Shark\\ndefiniti
;SETTING_3 on = prusa_i3\\n\\n[metadata]\\nsetting_version = 2\\nquality_type = 
;SETTING_3 normal\\ntype = quality_changes\\n\\n[values]\\nadhesion_type = brim\
;SETTING_3 \nbrim_width = 3.0\\ncool_fan_full_at_height = 0.5\\ncool_fan_speed_m
;SETTING_3 in = 75.0\\ngradual_infill_steps = 3\\nlayer_height = 0.15\\nlayer_he
;SETTING_3 ight_0 = 0.2\\nmaterial_bed_temperature = 45\\nmaterial_flow = 102\\n
;SETTING_3 material_print_temperature = 213\\nretraction_amount = 5\\nretraction
;SETTING_3 _speed = 30\\nskirt_line_count = 2\\nspeed_print = 50\\nsupport_enabl
;SETTING_3 e = True\\nsupport_type = buildplate\\n\\n"}

fieldOfView, 2017-09-27:
This is an architectural choice made in Cura.
PRs are welcome (either in Cura or in this plugin, but I would say this particular issue needs to be fixed in Cura)

Kriechi, 2018-10-06:
fixed by f1df408ebdadfbd21460833f2dfd0b15dca2d982

commit f1df408e (2018-10-06): Use gcodewriter plugin to compose gcode

Fixes #60, kudos to @Kriechi
commit 88d300d2 (2018-10-06): Use gcodewriter plugin to compose gcode

Fixes #60, kudos to @Kriechi