fieldOfView/Cura-OctoPrintPlugin #61: Error loading the plugin in Cura 2.7

kkalbaugh, 2017-10-06:
I am suddenly unable to load the plugin in cura.   Here's a dump of the log:

```
2017-10-06 13:52:10,949 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ConsoleLogger
2017-10-06 13:52:10,966 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: C:\Program Files\Cura 2.7\CuraEngine.exe
2017-10-06 13:52:10,967 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraEngineBackend
2017-10-06 13:52:10,974 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraProfileReader
2017-10-06 13:52:10,980 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraProfileWriter
2017-10-06 13:52:10,986 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJReader
2017-10-06 13:52:10,992 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJWriter
2017-10-06 13:52:11,003 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLReader
2017-10-06 13:52:11,010 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLWriter
2017-10-06 13:52:11,016 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger
2017-10-06 13:52:11,023 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeProfileReader
2017-10-06 13:52:11,033 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeReader
2017-10-06 13:52:11,040 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeWriter
2017-10-06 13:52:11,049 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ImageReader
2017-10-06 13:52:11,062 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LayerView
2017-10-06 13:52:11,071 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LegacyProfileReader
2017-10-06 13:52:11,081 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice
2017-10-06 13:52:11,090 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MachineSettingsAction
2017-10-06 13:52:11,135 - ERROR - UM.Logger.logException [76]: Exception: Error loading plugin OctoPrintPlugin:
2017-10-06 13:52:11,137 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-10-06 13:52:11,138 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\PluginRegistry.py", line 218, in loadPlugin
2017-10-06 13:52:11,139 - ERROR - UM.Logger.logException [80]:   File "C:\Program Files\Cura 2.7\plugins\OctoPrintPlugin\__init__.py", line 13, in register
2017-10-06 13:52:11,140 - ERROR - UM.Logger.logException [80]:     "output_device": OctoPrintOutputDevicePlugin.OctoPrintOutputDevicePlugin(),
2017-10-06 13:52:11,141 - ERROR - UM.Logger.logException [80]:   File "C:\Program Files\Cura 2.7\plugins\OctoPrintPlugin\OctoPrintOutputDevicePlugin.py", line 21, in __init__
2017-10-06 13:52:11,142 - ERROR - UM.Logger.logException [80]:     self._zero_conf = Zeroconf()
2017-10-06 13:52:11,143 - ERROR - UM.Logger.logException [80]:   File "zeroconf.py", line 1523, in __init__
2017-10-06 13:52:11,144 - ERROR - UM.Logger.logException [80]: OSError: [WinError 10022] An invalid argument was supplied
2017-10-06 13:52:11,154 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
2017-10-06 13:52:11,156 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PerObjectSettingsTool
2017-10-06 13:52:11,163 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PluginBrowser
2017-10-06 13:52:11,172 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PostProcessingPlugin
2017-10-06 13:52:11,186 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RemovableDriveOutputDevice
2017-10-06 13:52:11,232 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SliceInfoPlugin
2017-10-06 13:52:11,241 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SolidView
2017-10-06 13:52:11,250 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CameraTool
2017-10-06 13:52:11,260 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MirrorTool
2017-10-06 13:52:11,271 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RotateTool
2017-10-06 13:52:11,284 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ScaleTool
2017-10-06 13:52:11,292 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SelectionTool
2017-10-06 13:52:11,303 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin TranslateTool
2017-10-06 13:52:11,319 - DEBUG - UM.PluginRegistry.loadPlugin [199]: Plugin UltimakerMachineActions was disabled
2017-10-06 13:52:11,337 - DEBUG - UM.PluginRegistry.loadPlugin [199]: Plugin UM3NetworkPrinting was disabled
2017-10-06 13:52:11,349 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UpdateChecker
2017-10-06 13:52:11,351 - INFO - UpdateChecker.UpdateCheckerJob.run [44]: Checking for new version of cura
2017-10-06 13:52:11,380 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin USBPrinting
2017-10-06 13:52:11,392 - DEBUG - UM.PluginRegistry.loadPlugin [199]: Plugin VersionUpgrade21to22 was disabled
2017-10-06 13:52:11,399 - DEBUG - UM.PluginRegistry.loadPlugin [199]: Plugin VersionUpgrade22to24 was disabled
2017-10-06 13:52:11,409 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade25to26
2017-10-06 13:52:11,419 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade26to27
2017-10-06 13:52:11,428 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SimpleView
2017-10-06 13:52:11,442 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin X3DReader
2017-10-06 13:52:11,456 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XmlMaterialProfile
2017-10-06 13:52:11,466 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XRayView
2017-10-06 13:52:11,468 - INFO - UM.Qt.QtApplication.__init__ [112]: Command line arguments: {'qmljsdebugger': None, 'disable-textures': False, 'single_instance': False, 'external-backend': False, 'file': []}
2017-10-06 13:52:11,477 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-10-06 13:52:11,498 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening
2017-10-06 13:52:11,541 - INFO - UM.Backend.Backend.startEngine [75]: Started engine process: C:\Program Files\Cura 2.7\CuraEngine.exe
2017-10-06 13:52:11,543 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Calling engine with: ['C:\\Program Files\\Cura 2.7\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Cura 2.7\\resources\\definitions\\fdmprinter.def.json', '']
2017-10-06 13:52:11,620 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-06 13:52:11,622 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Cura_SteamEngine version DEV
2017-10-06 13:52:11,625 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Connected
2017-10-06 13:52:11,626 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Copyright (C) 2014 David Braam
2017-10-06 13:52:11,627 - DEBUG - UM.Backend.Backend._onSocketStateChanged [168]: Backend connected on port 49674
2017-10-06 13:52:11,628 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-06 13:52:11,630 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] This program is free software: you can redistribute it and/or modify
2017-10-06 13:52:11,632 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-10-06 13:52:11,633 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] the Free Software Foundation, either version 3 of the License, or
[... 20 lines cut ...]
2017-10-06 13:52:12,320 - WARNING - cura.MachineActionManager.addFirstStartAction [90]: Unable to add BedLevel to ultimaker_original_plus, as the action is not recognised
2017-10-06 13:52:12,354 - WARNING - cura.MachineActionManager.addSupportedAction [77]: Unable to add UpgradeFirmware to ultimaker2_plus, as the action is not recognised
2017-10-06 13:52:12,587 - WARNING - cura.MachineActionManager.addSupportedAction [77]: Unable to add DiscoverUM3Action to ultimaker3, as the action is not recognised
2017-10-06 13:52:12,588 - WARNING - cura.MachineActionManager.addFirstStartAction [90]: Unable to add DiscoverUM3Action to ultimaker3, as the action is not recognised
2017-10-06 13:52:13,137 - WARNING - cura.MachineActionManager.addSupportedAction [77]: Unable to add UpgradeFirmware to makeit_pro_l, as the action is not recognised
[... cut ...]
2017-10-06 13:52:14,741 - WARNING - cura.MachineActionManager.addSupportedAction [77]: Unable to add DiscoverUM3Action to ultimaker3_extended, as the action is not recognised
2017-10-06 13:52:14,742 - WARNING - cura.MachineActionManager.addFirstStartAction [90]: Unable to add DiscoverUM3Action to ultimaker3_extended, as the action is not recognised
2017-10-06 13:52:14,839 - WARNING - cura.MachineActionManager.addSupportedAction [77]: Unable to add UpgradeFirmware to tam, as the action is not recognised
```

averter, 2017-10-07:
I am also unable to use the plugin in Cura 2.7. Since I am using the PPA version and the label says it's for the 2.6 version I assume it hasn't been updated yet.

fieldOfView, 2017-10-07:
@averter, please use the official AppImage instead of the PPA. The PPA has never been officially supported, and has caused some issues recently because of it. This does not help @kkalbaugh though, because he seems to be running Windows.

@kkalbaugh, you seem to have disabled a couple of plugins, correct? Anything else you have done to your installation that I should know about? There seems to be an issue on your installation with the external module we use for discovery of OctoPrint instances which I have never seen before. I will make the plugin more robust for this, so if the discovery fails you can still use the plugin.

kkalbaugh, 2017-10-08:
I didn't disable anything.  I've uninstalled/ reinstalled Cura twice and deleted all the %appdata% left behind, but still get the issue.  Running Windows 10 1703, if that helps any.

fieldOfView, 2017-10-08:
At some point, you did:
```
2017-10-06 13:52:11,319 - DEBUG - UM.PluginRegistry.loadPlugin [199]: Plugin UltimakerMachineActions was disabled
2017-10-06 13:52:11,337 - DEBUG - UM.PluginRegistry.loadPlugin [199]: Plugin UM3NetworkPrinting was disabled
```
Note that if you remove the entire cura folder from %appdata% (~\AppData\Roaming\cura), Cura will try to find a previous version folder from %localappdata% (~\AppData\Local\cura). So you can still end up with old configuration. Please try leaving an empty folder named "cura" in place if you want to try starting over from scratch.

fieldOfView, 2017-10-18:
Sorry, this dropped off my radar for the 3.0 release :-/
Fixed here: cc9d0aea0cfcbdabbb7324ca792e3b416760ab3a

kkalbaugh, 2017-10-18:
Thank you @fieldOfView !

averter, 2017-10-18:
For some reason I am unable to use the official Appimage. I follow all the installation steps and when I open the printers manager the octoprint button does not show up. No wonder that I tend towards the PPA.

fieldOfView, 2017-10-18:
@Averter, that is probably caused by this bug. I'll look in to a workaround for you.

fieldOfView, 2017-10-31:
This should be fixed in Cura 3.0.4

closed 2017-10-31 by fieldOfView