fieldOfView/Cura-OctoPrintPlugin #62: Plugin stopped loading related to Zeroconf

jamesgol, 2017-10-17:
I don't believe this is the same issue as #61, but it seems similar (maybe just the OS difference gives a different error)

```
2017-10-17 11:50:04,227 - ERROR - UM.Logger.logException [76]: Exception: Error loading plugin OctoPrintPlugin:
2017-10-17 11:50:04,229 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-10-17 11:50:04,231 - ERROR - UM.Logger.logException [80]:   File "/build/3.0/build/inst/lib/python3.5/site-packages/UM/PluginRegistry.py", line 225, in loadPlugin
2017-10-17 11:50:04,232 - ERROR - UM.Logger.logException [80]:   File "/tmp/.mount_tgU5wM/usr/bin/plugins/plugins/OctoPrintPlugin/__init__.py", line 13, in register
2017-10-17 11:50:04,234 - ERROR - UM.Logger.logException [80]:     "output_device": OctoPrintOutputDevicePlugin.OctoPrintOutputDevicePlugin(),
2017-10-17 11:50:04,235 - ERROR - UM.Logger.logException [80]:   File "/tmp/.mount_tgU5wM/usr/bin/plugins/plugins/OctoPrintPlugin/OctoPrintOutputDevicePlugin.py", line 21, in __init__
2017-10-17 11:50:04,237 - ERROR - UM.Logger.logException [80]:     self._zero_conf = Zeroconf()
2017-10-17 11:50:04,238 - ERROR - UM.Logger.logException [80]:   File "/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1523, in __init__
2017-10-17 11:50:04,240 - ERROR - UM.Logger.logException [80]: OSError: [Errno 105] No buffer space available
```
This started happening a few weeks ago using 2.7.0, but it definitely did work for a while with 2.7.0.  I just tried upgrading to 3.0.3 to see if there were any improvements but no luck.

I've tried upgrading everything, rebooting, removing all cura related files I could find (~/.config/cura and ~/.local/share/cura).  Even tested back with 2.6.2 and get the same results.

I'm almost certain this is not exactly a problem with the plugin, but something related to the kernel/OS.  If I figure out the issue I'll update here in case it will help someone else.

fieldOfView, 2017-10-18:
Fixed here: c9d0aea0cfcbdabbb7324ca792e3b416760ab3a

Unfortunately this was fixed after the release of Cura 3.0, so a release will have to wait 2 months. If you want to fix it yourself in the mean time, you can download this file:
https://github.com/fieldOfView/OctoPrintPlugin/raw/cc9d0aea0cfcbdabbb7324ca792e3b416760ab3a/OctoPrintOutputDevicePlugin.py
Replace the file in Program Files\Cura 3.0\plugins\OctoPrintPlugin with this one.

closed 2017-10-18 by fieldOfView

jamesgol, 2017-10-18:
I'll give the fix a try, though I'm confused as to why everything worked fine in older versions before but doesn't work now even when trying the older versions.  Nothing on my network has changed, though my linux kernel version has been updated and I haven't tried rolling that back yet.

fieldOfView, 2017-10-18:
The fix just handles the issue more gracefully. Auto discovery will not work if ZeroConf does not work.