fieldOfView/Cura-OctoPrintPlugin #64: Cura 3.1 crashes when trying to connect to Octoprint

drayde, 2018-01-17:
Inside Cura 3.1 (Win10 64bit) when I try to connect to OctoPrint, I get an exception:

```
2018-01-17 18:35:54,828 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction.testApiKey [141]: Trying to access OctoPrint instance at http://192.168.178.61:80/ with the provided API key.
2018-01-17 18:35:58,422 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction._onRequestFinished [297]: API key accepted by OctoPrint.
2018-01-17 18:36:02,645 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-01-17 18:36:02,652 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-01-17 18:36:02,659 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "C:\Program Files\Ultimaker Cura 3.1\plugins\OctoPrintPlugin\DiscoverOctoPrintAction.py", line 227, in applyGcodeFlavorFix
2018-01-17 18:36:02,665 - CRITICAL - cura.CrashHandler.__init__ [65]:     definition_changes_container = CuraStackBuilder.createDefinitionChangesContainer(
2018-01-17 18:36:02,671 - CRITICAL - cura.CrashHandler.__init__ [65]: NameError: name 'CuraStackBuilder' is not defined
```

This happens both with the plugin version that comes with Cura 3.1 as well as the latest from the 3.1 branch

fieldOfView, 2018-01-17:
Thanks for reporting, I will get that fixed for Cura 3.2.

As a workaround, you can go into Machine Settings and set gcode flavor to "Marlin" before connecting to OctoPrint.

drayde, 2018-01-17:
That worked! Thanks for the quick reply!

closed 2018-01-28 by fieldOfView