fieldOfView/Cura-OctoPrintPlugin #65: OctoPrint 3.2.1 Pluggin

RobertsClif, 2018-01-27:
Installed the OctoPrint 3.2.1 Pluggin into Cura 3.2 Beta I only have control over heating, XYZ Axis Controls home features, I can see the temperature change. If I try to sent a sliced model to OctoPrint It goes thru the motions but does not send the sliced model, As stated previous Cura 3.1 work Perfect.

Am I missing something

Thank You

NearlCrews, 2018-01-27:
Exact same problem here. Cura connects, has control of the printer, but doesn't send any gcode to OctoPrint.

fieldOfView, 2018-01-27:
Could you post your logs somewhere so I can have a look?

> Am I missing something

Beyond that Cura 3.2 is a beta, and thing can break with beta software?

The decision to make OctoPrint an optional plugin again (instead of installing it by default) wasn't really thought through enough, and as a result some things went "less than optimal". Currently you cannot even download the plugin anymore through the Plugin Browser (which I am not very happy with). It is also impossible for me to know which exact version of the plugin you are currently using.

Could you try http://files.fieldofview.com/temp/OctoPrintPlugin.curaplugin

RobertsClif, 2018-01-27:
FYI cannot find a way to change the internal OctoPrintPlugin with OctoPrintPlugin.curaplugin above
And I did find this OctoPrintPlugin-3.2.zip but also could not find a way to remove the internal Pluggin.
[OctoPrintPlugin-3.2.zip](https://github.com/fieldOfView/OctoPrintPlugin/files/1670180/OctoPrintPlugin-3.2.zip)

Log file
[octoprint.log](https://github.com/fieldOfView/OctoPrintPlugin/files/1670173/octoprint.log)

fieldOfView, 2018-01-27:
In Cura 3.2, there is no internal octoprint plugin. Please use the link I posted above to a curaplugin, and drag that into Cura. You will have to restart Cura after doing so.

Thanks for the logs, though I meant the logs for Cura (these are the logs of your octoprint server). You can find the cura logs by starting Cura and going to Help -> Show configuration folder. There should be a cura.log in the folder that opens.

NearlCrews, 2018-01-27:
I just tried the linked curaplugin file above. It loads without any issue on Cura 3.2.0-BETA, but I'm still not seeing anything in Octoprint. This is the only relevant error in the log:

2018-01-27 16:48:08,297 - ERROR - OctoPrintPlugin.OctoPrintOutputDevice._startPrint [485]: An exception occurred in network connection: name 'global_container_stack' is not defined

Other than that it's just the usual log messages. I get that error right after the UI says it's sending data to Octoprint. I also agree 100% about their decision to remove the plugin not really being thought through. It's incredibly useful.

fieldOfView, 2018-01-27:
Thanks, that helps. I have uploaded a new version that should fix this. Same link as above.

NearlCrews, 2018-01-28:
Awesome, that seems to have fixed it. The gcode is showing up as expected. Thanks much and keep up the great work. Hopefully we'll see this included with Cura again.

RobertsClif, 2018-01-29:
Not sure why this was closed still have an issue

Video shows OctoPrintPlugin.curaplugin (3.2.1) failing to send the sliced model to Cura 3.2 Beta.
https://youtu.be/rQG9IaEsIDg

fieldOfView, 2018-01-29:
The issue was closed because @NearlCrews seemed to have the same issue, and because he gave me logs I could identify the problem and fix it. I thought you were affected by the same issue.

The logs you gave me are fairly useless to me, because they are the octoprint server logs; I need the Cura logs. The video is somewhat hard to follow, because the mousepointer is not visible. The print seems to start just fine, though it does not wait for the nozzle to preheat. The latter may indicate something wrong with your start gcode? Perhaps if you tell me what is going wrong in the video, I'll have more of a clue.

RobertsClif, 2018-01-29:
Is this the fioe you wanted

[cura.log](https://github.com/fieldOfView/OctoPrintPlugin/files/1675498/cura.log)

I uninstall Cura 3.2 Beta Completely, the video shows everything I did on htis log.

fieldOfView, 2018-01-29:
Thanks for the log. The error that is in the log is the same error that @NearlCrews reported (https://github.com/fieldOfView/OctoPrintPlugin/issues/65#issuecomment-361018056) after my initial fix. I uploaded a new version of the plugin which fixes that bug.

Please redownload the plugin here and try installing it again:
http://files.fieldofview.com/temp/OctoPrintPlugin.curaplugin

fieldOfView, 2018-02-03:
@RobertsClif, did you have time to retest the plugin?

RobertsClif, 2018-02-03:
I had re-downloaded the plugin 7 days ago but it did not work then.

I downloaded it again and it appears to function correctly.