fieldOfView/Cura-OctoPrintPlugin #66: HTTPS libs missing

Phaiax, 2018-01-29:
Hey

I have an issue where I am not sure if the plugin could resolve this.

I use cura 3.1 on fedora, but start it using the AppImage, so I think it is kind of sandboxed.

If I try to connect to octoprint on http, it connects and accepts the API key.

If I switch to HTTPS, I get these in the console:
```
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot resolve ERR_free_strings
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_cleanup
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_init
qt.network.ssl: QSslSocket: cannot resolve sk_new_null
qt.network.ssl: QSslSocket: cannot resolve sk_push
qt.network.ssl: QSslSocket: cannot resolve sk_free
qt.network.ssl: QSslSocket: cannot resolve sk_num
qt.network.ssl: QSslSocket: cannot resolve sk_pop_free
qt.network.ssl: QSslSocket: cannot resolve sk_value
qt.network.ssl: QSslSocket: cannot resolve SSL_library_init
qt.network.ssl: QSslSocket: cannot resolve SSL_load_error_strings
qt.network.ssl: QSslSocket: cannot resolve SSL_get_ex_new_index
qt.network.ssl: QSslSocket: cannot resolve SSLv2_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv2_server_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_server_method
qt.network.ssl: QSslSocket: cannot resolve X509_STORE_CTX_get_chain
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf
qt.network.ssl: QSslSocket: cannot resolve SSLeay
qt.network.ssl: QSslSocket: cannot resolve SSLeay_version
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function sk_num
file:///run/media/daniel/AppImage/usr/bin/plugins/plugins/OctoPrintPlugin/DiscoverOctoPrintAction.qml:236: TypeError: Cannot read property 'baseURL' of undefined
file:///run/media/daniel/AppImage/usr/bin/plugins/plugins/OctoPrintPlugin/DiscoverOctoPrintAction.qml:236: TypeError: Cannot read property 'baseURL' of undefined
file:///run/media/daniel/AppImage/usr/bin/plugins/plugins/OctoPrintPlugin/MonitorItem.qml:5:5: QML Image: Failed to get image from provider: image://camera/2
file:///run/media/daniel/AppImage/usr/bin/resources/qml/Preferences/MachinesPage.qml:164: TypeError: Cannot read property 'connectionText' of null
file:///run/media/daniel/AppImage/usr/bin/plugins/plugins/OctoPrintPlugin/DiscoverOctoPrintAction.qml:236: TypeError: Cannot read property 'baseURL' of null
file:///run/media/daniel/AppImage/usr/bin/resources/qml/Preferences/MachinesPage.qml:118: TypeError: Cannot read property 'onCompleted' of null
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
```

The last two lines every time I try to connect.

Any idea how to insert the ssl libs? I'll go on research myself.

fieldOfView, 2018-01-30:
Thanks for the report. I don't think I can fix this on my end, but I will log an issue with the Cura project.
If you want to fix this locally, you would have to "unpack" the appimage (mount the image, copy the contents somewhere), and add/replace the libs (but I don't know what libs you would have to add).

fieldOfView, 2018-08-15:
I think this is finally being fixed in the next version of Cura (because other plugins like the cloud backup also need SSL connections)

fieldOfView, 2018-09-30:
Can you test if this is fixed in Cura 3.5 beta?