fieldOfView/Cura-OctoPrintPlugin #70: Plugin shows 100% completion when pre-heating is active

rogerwlucas, 2018-03-14:
Cura 3.2.1 on Linux with Octoprint 1.3.6.

When you submit a print to Octoprint from Cura, it takes time before Octoprint reports meaningful estimates.  During the pre-heat phase, Octoprint reports zero for printing time and estimated time left but the print hasn't started.  At this point, the plugin reports 100% print progress as shown below.

I can see why it might calculate this.... but it's not right.

As soon as the print starts, the print progress calculation updates to a more sensible value.

[image]

rogerwlucas, 2018-03-15:
I suspect this is because Octoprint reports "null" for a lot of values during the pre-heat phase and the progress indicator isn't updated if the completion value doesn't evaluate to true.

The patch below handles the case when the completion value is null and ensures that the progress is always updated:

~~~~
--- OctoPrintOutputDevice.py.orig       2018-03-14 16:57:36.263690948 +0000
+++ OctoPrintOutputDevice.py    2018-03-15 10:28:22.794357634 +0000                                                                                                                                                                                                            
@@ -704,9 +704,8 @@                                                                                                                                                                                                                                                            
                         Logger.log("w", "Received invalid JSON from octoprint instance.")                                                                                                                                                                                     
                         json_data = {}                                                                                                                                                                                                                                        
                                                                                                                                                                                                                                                                               
-                    progress = json_data["progress"]["completion"]                                                                                                                                                                                                            
-                    if progress:                                                                                                                                                                                                                                              
-                        self.setProgress(progress)                                                                                                                                                                                                                            
+                    progress = json_data["progress"]["completion"] or 0                                                                                                                                                                                                       
+                    self.setProgress(progress)                                                                                                                                                                                                                                
                                                                                                                                                                                                                                                                               
                     if json_data["progress"]["printTime"]:                                                                                                                                                                                                                    
                         self.setTimeElapsed(json_data["progress"]["printTime"])                                                                                                                                                                                               
~~~~

fieldOfView, 2018-03-15:
Thanks. Note that the OctoPrintPlugin is a 3rd party plugin (by me), and its issue queue is here:
https://github.com/fieldofview/octoprintplugin

fieldOfView, 2018-03-15:
This has already been fixed in a different way in master, the version that will be compatible with Cura 3.3

rogerwlucas, 2018-03-15:
Hi Aldo, thanks for the update.  I love the plugin BTW - I wouldn't have taken the effort to report the issues and go digging if I didn't!  :-)

closed 2018-03-15 by rogerwlucas