fieldOfView/Cura-OctoPrintPlugin #71: Duet Wifi extension

cbusillo, 2018-03-18:
Hi!  I love this plugin.  Is there any chance it can be extended to control a Duet Wifi board?  Thanks.

fieldOfView, 2018-03-18:
The protocol for the Duet (Ethernet or Wifi) is so different that it would make more sense to make a separate plugin for it. I don't currently have a Duet board, so that would make it hard to implement this.

cbusillo, 2018-03-19:
Thanks for the feedback.  I had hoped it was similar.  The available Duet plugin was forked from a different OctoPrint plugin.  It works, but your design is so much better =). I might have to get the Duet guys I know to chip in on a board for you!

fieldOfView, 2018-03-19:
> The available Duet plugin was forked from a different OctoPrint plugin.

Perhaps you have a link?

cbusillo, 2018-03-19:
Duet plugin for cura
https://github.com/Kriechi/Cura-DuetRRFPlugin

Based on octoprint plugin for Cura 
https://github.com/markwal/Cura-OctoPrintUpload

Issue on DuetRRFPlugin about merging
https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/6

fieldOfView, 2018-03-19:
That plugin looks fairly comprehensive to me. The part that is missing in the plugin is also the part that I would not know how to implement without access to the hardware; I can't test polling the state of the printer.

For now I am going to close this issue.

cbusillo, 2018-03-19:
Sure thing!  Thanks for taking a look.  I use your plugin on my one printer still =). If you ever change your mind let me know, I can at least produce the http stuff.

closed 2018-03-20 by fieldOfView