fieldOfView/Cura-OctoPrintPlugin #73: [Cura 3.3] Seems to detect printer, but cannot click "Print with OctoPrint"

probonopd, 2018-04-10:
Trying to use my printer with WirelessPrinting (which implements a subset of OctoPrint API) with Cura 3.3. It seems to find the printer:

[image]

But when I click "Print with OctoPrint" nothing happens at all - no output on the command line either. How can I debug this?

probonopd, 2018-04-10:
OK, it seems like what used to be printed to the command line is now hidden away in `$HOME/.local/share/cura/3.3/cura.log`.

```
$ cat ./.local/share/cura/3.3/cura.log
2018-04-10 14:21:23,813 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/home/me/.local/share/cura/3.3/plugins/OctoPrintPlugin/OctoPrintPlugin/OctoPrintOutputDevice.py", line 302, in requestWrite
2018-04-10 14:21:23,815 - ERROR - [MainThread] UM.Logger.logException [85]:     active_build_plate = Application.getInstance().getBuildPlateModel().activeBuildPlate
2018-04-10 14:21:23,817 - ERROR - [MainThread] UM.Logger.logException [85]: AttributeError: 'BuildPlateModel' object has no attribute 'activeBuildPlate'
```

closed 2018-04-10 by probonopd

fieldOfView, 2018-04-10:
There should be an updated version of the plugin in the plugin browser that fixes this issue (see https://github.com/fieldOfView/OctoPrintPlugin/commit/49e30e3e6225bd304f0d7a512c45053856add9a2)

fieldOfView, 2018-04-10:
In Cura 3.2 and 3.3 no OctoPrintPlugin comes with Cura. You have to install it manually.

The plugin browser should show which version is available (3.3.1 currently), I think you have 3.3.0.

probonopd, 2018-04-10:
> You have to install it manually.

I have selected it in the plugin browser, yes. Does this download it on demand? That was not clear to me as a user. I thought it came with Cura and just had to be activated. I much preferred the "old way" of it working out-of-the-box...

probonopd, 2018-04-10:
jfwiw, I don't see an "Update" button...

fieldOfView, 2018-04-10:
Please open an issue in the Cura tracker, there is little I can do about it (other than fixing their bugs, which I've done enough for a bit).

You can probably "uninstall" (remove?) the plugin and reinstall it.

fieldOfView, 2018-04-11:
Did uninstalling and installing work?

probonopd, 2018-04-11:
Define "work" - I have another issue at the moment; namely that the receiving end of this crashes when uploaded through Cura (for reasons that I still need to debug).

fieldOfView, 2018-04-11:
Ok, but that is not the issue you describe above, and you said it also happens now on 2.7, so I’m blaming your code ;)