fieldOfView/Cura-OctoPrintPlugin #76: Octoprint webcam not visible in Cura 3.3

Galfinite, 2018-04-24:
Webcam view is working fine in Cura 3.2.1, but without change to the Octoprint, the webcam does not work in 3.3. I can have both program versions open side by side, and one works, the other doesn't. The printer's active info does update correctly, only the webcam is at issue.

edasque, 2018-05-03:
I feel that the upload is much slower too, is it just me?

fieldOfView, 2018-05-03:
@edasque: that would be a separate issue.

fieldOfView, 2018-05-03:
See also https://github.com/Ultimaker/Cura/issues/3725

VMaxx, 2018-05-07:
I've used your code to make the Repetier plugin and I the same problem there.  I got it to come up once, but subsequent loads it was gone and I've not gotten it back since.

fieldOfView, 2018-06-11:
Could you test the fix here?
https://github.com/Ultimaker/Cura/issues/3725#issuecomment-396181967

fieldOfView, 2018-06-11:
Fix seems to work. A new version of the plugin has been submitted.

closed 2018-06-11 by fieldOfView

VMaxx, 2018-06-21:
Thanks, this helped me fix the Repetier plugin too.