fieldOfView/Cura-OctoPrintPlugin #79: Allow Print with OctoPrint to power-on printer

posicat, 2018-05-25:
I've got my printer setup so that my OctoPi controls a relay to power the entire printer on or off, so normally my printer is powered off.  Than means if I hit Print with OctoPrint" I get an error that the printer is not on (but OctoPi is turned on, waiting for a command to turn the printer on).   If there were any way for the plugin to detect that PSU Control is installed and send the command to turn the printer on, that would be amazing.

This is the plugin I'm using : https://plugins.octoprint.org/plugins/psucontrol/

fieldOfView, 2018-08-15:
How long does it typically take for your printer to be connected and "ready" after it is turned on with the plugin?

posicat, 2018-08-15:
I clicked the power icon, and after 15 seconds the connect button, and it took about 15 more seconds to finish the connection.  Round both to 20 for a default value probably.

fieldOfView, 2018-08-16:
So you still have to manually connect? Hmm, that's not awesome. Fortunately I can issue a "connect" command through the API.

posicat, 2018-08-16:
There's another plugin that should be doing the auto-connect, that's not at
the moment (https://github.com/markwal/OctoPrint-PortLister) that I'm
hoping isn't a dead project since I don't know Python very well.

I guess the ideal would be if your plugin knew the port-name the printer
should be on, and would select that and click connect once it's powered on.
It would removed dependence on the PortLister plugin, and only rely on the
PSU plugin.

posicat, 2018-08-22:
I wonder if it might not be a good idea to have a plugin specifically for handling Cura Startup commands, that other plugins could hook into to listen for events.  Some kind of a common interface for : Is printer On? Can printer be turned on?   Is printer connected?  Can printer be connected?  Etc.  And it can be aware of other plugins at the moment, but eventually will expect the other plugins to grow up and use the hooks it defines.

chuckdaniel, 2018-12-16:
I use the TP Link plug-in and it works great for exactly this.    Just set it up and add the M80 command w/IP to the preprocessing in Octoprint.

Tailslide, 2019-03-11:
I  would love to see this feature as well.. PSU Control has an API for querying and powering on. There is no workaround for me since I use the rambo board that does not boot until it gets power from the power supply (it will not run off USB power).  I have a batch file that I use with simplify 3d to query and call the API to power on the printer then wait until it is connected to if anyone needs it.

bullitt186, 2019-05-27:
I'd also love to see that feature.

I use MQTT to power on my PSU (controlled via a MQTT-controlled relais). Currently i am triggering the MQTT command via the MQTTPublish plugin by sending a custom @MQTTPublish message from the GCODE. After the PSU is on, i use the PortLister plugin for auto-connecting to the printer once it's serial port is available.

So as soon as the CURA OctoPrint Plugin would allow me to start Printing withou checking the status of the printer first, I'd be good to go. It would look like this:

- Start GCODE in CURA for my Printer containts MQTT command to turn on PSU
- On clicking "Print with Octoprint", Cura Octoprint Plugin sends Print Job to OctoPrint
- OctoPrint starts executing GCODE
- GCODE starts with MQTT command to turn on PSU
- GCODE waits a minute for PSU to come up and for PortLister to find the newly available port and connect
- After a minute, Regular Print Job GCODE starts as normal

What's hindering this, is that the Cura Octoprint plugin checks the connection status of the printer.
So can you please make a check box "start printing w/o checking printer connection status"?

Thanks a lot for the great plugin!

chriswal, 2019-08-08:
as i understand octoprint the printer must be online that gcode commands are processed. thats why i used the PSUControl plugin to switch on the printer. I have a fork with autoconnect. Also i created a fork of the octoprint plugin which can turn on the printer and wait 20 sec its the 3.5 branch tested with cura 4.2.1

fieldOfView, 2019-08-12:
Initial support for turning on the printer with PSU Control is now in the 3.5 branch, which will be the next release of the plugin. Thanks @chriswal

closed 2019-08-12 by fieldOfView

chriswal, 2019-08-12:
There must be some sytax error ist not loading the Plugin

fieldOfView, 2019-08-12:
Thanks for noticing. Should be fixed now. I don't have the PSU Control plugin installed though, so please test the functionality and let me know if it (still) works after my changes.

chriswal, 2019-08-12:
I checked the power on part ..this is working ..
The hiding of the settings i will test a later time

bergsm, 2019-12-28:
Any way to make this work with the [TP-Link Smartplug Plugin](https://plugins.octoprint.org/plugins/tplinksmartplug/)? It seems like it would be pretty similar to the PSU Control. The TP-Link plugin itself is actually inspired by the PSU plugin.

fieldOfView, 2019-12-29:
@bergsm, can you open a new issue for this?

bergsm, 2019-12-29:
No problem! See #141.

drewzh, 2020-01-06:
This support seems to be broken in a recent version change. At least, when clicking Print in Cura, the printer turns out (great), the view switches to the Camera view, but the print never started. My Octoprint is setup to automatically connect and if I navigate to Octoprint after hitting print, the printer is indeed turned on and connected to Octoprint. This was working fine for quite some time and has only recently stopped working. Thanks FYI @fieldOfView

fieldOfView, 2020-01-06:
When your printer is turned on, does it also automatically connect in OctoPrint? How long does it take between getting turned on and the printer being connected in OctoPrint?

drewzh, 2020-01-07:
@fieldOfView Hi, it generally takes up to 20 seconds to automatically connect. I use the auto connect plugin (PortLister) to automatically connect when the serial port sees activity. Up until recently, this new integration has been such a godsend - I can't wait to get it going again :)

fieldOfView, 2020-01-07:
Could you actually time it? Currently, there's a 20 second delay between telling PSU Control to turn on the printer and the plugin continuing to start the print. I may end up having to make that time configurable.

drewzh, 2020-01-07:
@fieldOfView I just timed the last one at 26 seconds. The previous one was around the 20 mark. Is there no provision for querying the status before issuing the print command? I'm not familiar with the API myself.

fieldOfView, 2020-01-07:
> Is there no provision for querying the status before issuing the print command?

When should it give up though?

I'll see what I can do. I'm looking at improving that code anyway, because of #141. 
As a quick fix, you could try increasing the timeout delay in turnOnPrinter here:
https://github.com/fieldofview/OctoPrintPlugin/blob/80aca632841024882b6e295f34f28cf7358793bf/OctoPrintOutputDevice.py#L341
Would be nice if you could let me know if that fixes the problem.

fieldOfView, 2020-01-07:
Sorry, that code has been refactored, it now lives here:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/38d53a36e408f9cecd56048d19a7635516477b38/OctoPrintOutputDevice.py#L168

drewzh, 2020-01-08:
> When should it give up though?
>
> I'll see what I can do. I'm looking at improving that code anyway, because of #141.
> As a quick fix, you could try increasing the timeout delay in turnOnPrinter here: [...]

I just came across the following commit that was recently added to Octoprint PSU control plugin (do you interface with this or via some other means?): https://github.com/kantlivelong/OctoPrint-PSUControl/pull/141/commits/d49477ebb482c08602cfab6e8ca13bf5839b0172

drewzh, 2020-01-08:
> Sorry, that code has been refactored, it now lives here:
> https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/38d53a36e408f9cecd56048d19a7635516477b38/OctoPrintOutputDevice.py#L168

Honestly I'm quite busy with other things at the moment so I'm happy to wait - it's not super important but is very handy. I think using an active poll to find the state of the printer before printing (and then timing out after X seconds) is a more desirable behaviour as each printer can take a non-predeterminate amount of time to startup and this would provide more compatibility.

fieldOfView, 2020-01-09:
> I just came across the following commit...

That's about whether or not the PSU has been turned on, not about wheter or not the printer is connected to OctoPrint.

I'm solving this issue now by periodically checking if the printer has become available after turning on the PSU, instead of using a fixed timer to delay a print after turning on the PSU. While waiting, you get the option to stop waiting and queue the print, or to cancel the print altogether.

drewzh, 2020-01-09:
Of course! I wasn't thinking clearly.

That's great - looking forward to testing. I'll keep an eye out :)

fieldOfView, 2020-01-12:
I've been working on support for the following plugins for automatically turning the printer on when uploading a print: TP-Link Smartplug, Orvibo S20, Wemo Switch, Tuya Smartplug, Domoticz, MyStrom Switch, Tasmota and Tasmota MQTT (along with PSUControl, which was already supported). One of these plugins is detected, the Connect to OctoPrint dialog will have an option to "Automatically turn on the printer" using a configured plug. After sending the API command, the plugin will wait until the printer is connected to OctoPrint and continue uploading the file.

Here's a development snapshot of the plugin: [CuraPackage for Cura 4.4.x](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/UW44dINkkhO5BDyONGJIr9M1_GsYLNbEAeSAjNp3YJgG/OctoPrintPlugin-v7.0.0-2020-01-12T14%3A49%3A28Z.curapackage?Expires=1894200572&GoogleAccessId=stardust-storage-production%40stardust-193112.iam.gserviceaccount.com&Signature=KGx%2BV9VGQHbbS2U6%2B9J5bBSscSHg5KzQU5ZSCBjqkHrTvOH5BRnQLKKHgPJBYDWXPnae8m%2BcVkLanroqoD%2Bgo2OzQBU4l8RZe88Qyjf1G6b8%2FcKria8nxXjEukOnYq3JcPw2CZspuvx3aiJLnXcwb4cJn3W4CIR6%2FZCUK%2BtdsOBQSMLvZxmHiggoNdRDOX4f2zIk%2FRi8c0HLFnLfIzcPLRCkPMTUttvchLK3Gr9CZSKdcgTaQwY1f4lx1N2czaS0GrblBjUSR3ZuzoNKhdGiIYhecq64B7JaUvdm8c4JISpvesExSCpRbRQ0BPxJ2Nr8Y9L7OL%2FPpNGLIwIUIpQGPg%3D%3D). Download the file, and drop it into the running Cura application window (not into a folder!). Then restart Cura.

If you used the PSU Control support before, you may have to visit the "Connect to OctoPrint" dialog again to reset the option.