fieldOfView/Cura-OctoPrintPlugin #80: Plugin repeatedly needs "Update"

jameswood, 2018-05-31:
This plugin is great and has made my workflow super-satisfying.

I am having a problem where the plugin will just stop running. Cura's plugin manager then shows options to Uninstall or Update. Updating fixes the problem (after a Cura restart) for a session, but eventually the plugin will stop again and require another "update" and Cura restart.

[image]

It's not a huge problem, and I think it's probably a Cura problem rather than a plugin problem, but I wondered if anyone had any clues about what I should check?

Thanks for any advice!

fieldOfView, 2018-05-31:
That is strange, and the first time I hear of this issue.
Could you share the cura.log from your configuration folder? It is probably too large to attach here, so use a service like dropbox or another cloud drive and post a link.

jameswood, 2018-06-05:
Heyo! Sorry for the delay, as per the standard procedure the problem didn't appear again for some time.

Here are two cura.logs: https://we.tl/bM4zsO4TBq

One was the active log when I first noticed the problem. I've named that one `duringFailure-cura.log`

The other was a fresh log made immediately after the problem reappeared. I have not yet "updated" the plugin. It's called `afterFailure-cura.log`.

Thanks for any help/direction you can give me!

fieldOfView, 2018-06-05:
Could you upload your cura.cfg file?

fieldOfView, 2018-06-05:
It looks like you have disabled the plugin in a previous version, and the current version may not let you "undisable" it. Editing the cura.cfg file (once) should help.

jameswood, 2018-06-06:
I'd rather not upload it because there are passwords, paths, etc in it - but I can see that it includes the line `disabled_plugins = ,OctoPrintPlugin` so I'll delete that and see if it helps. Is there anything else in `cura.cfg` that I should look for?

fieldOfView, 2018-06-06:
No, just remove that line.

closed 2018-06-11 by fieldOfView

jameswood, 2018-06-12:
Thank you so much for your help - this fixed it. Sorry for not getting back to you sooner, I thought it was best to make sure the fix stuck, which it did!