fieldOfView/Cura-OctoPrintPlugin #83: Report errors better

lf-, 2018-08-14:
So I tried to upload a file which was blocked by my HTTP server for being too large. This issue manifested itself as clicking the upload button, successful-looking results appearing in the messages window, and the file not appearing in OctoPrint. The specific error code returned by my HTTP server which the plugin ignored was a 413 "Request Entity Too Large".

fieldOfView, 2018-08-15:
Good point, though as far as I know OctoPrint does not issue HTTP 413 responses; are you using another server as a reverse proxy?

fieldOfView, 2018-08-15:
Note that I do agree that showing an error to the user is better than failing silently. I’m working on that in the master branch.

lf-, 2018-08-15:
You are 100% correct that the actual source of the error was nginx. I fixed that after OctoPrint provided a similarly useless error accusing me of using the wrong file type but including an iframe of the actual error.

fieldOfView, 2018-09-17:
This has been implemented in the release for Cura 3.5

closed 2018-09-17 by fieldOfView