fieldOfView/Cura-OctoPrintPlugin #84: Uploads don't upload if printer is offline

lf-, 2018-08-14:
Use case: printer is offline for whatever reason but I want to slice the job immediately and print it once the printer is ready. Plugin is configured to *not* "automatically start print job after uploading", so it should be able to just upload it without the printer being online. Actually, this feature could be fleshed out more as well to change the print button text to "Upload to OctoPrint" instead of "Print with OctoPrint".

fieldOfView, 2018-08-15:
> it should be able to just upload it without the printer being online

I'll see if I can fix that by skipping the Queue Job button thing when the plugin is configured to not print automatically, for the next release.

> change the print button text to "Upload to OctoPrint" instead of "Print with OctoPrint".

Unfortunately, that is not possible from the plugin; it would require changes in Cura.

fieldOfView, 2018-08-15:
Main issue should be fixed in the next release

closed 2018-08-15 by fieldOfView