fieldOfView/Cura-OctoPrintPlugin #85: Plugin won't install

scratchyone, 2018-08-15:
I'm not sure if this is a cura issue, but I can't install this plugin. I installed it through cura and restarted, but under "installed" it's unchecked, so I check it and restart cura. I reopen the installed window and it's unchecked again. I uninstall the plugin and install from source. Still having the same issue. I also completely reinstalled cura and it won't work. I'm using cura 3.4.1. It shows up as unchecked and grayed out.

scratchyone, 2018-08-15:
Resetting my cura config fixed it

closed 2018-08-15 by scratchyone

fieldOfView, 2018-08-15:
When a plugin is unchecked in the "installed" tab, that means the plugin is installed but disabled. You can enable it by checking the checkbox and restarting Cura

scratchyone, 2018-08-15:
Thanks. I think my cura config had settings for the old version because every time I checked it and restarted it would be unchecked. I (accidentally) reset my cura config while trying to change the temp for PLA, and when I checked the box, this time it actually stayed,

bergsm, 2019-12-25:
I am having this issue with Cura 4.4.0, however even after wiping my cura configuration, I'm still unable to enable the OctoPrint Connection plugin

fieldOfView, 2019-12-26:
Please post a link to your cura.log (Help -> Show configuration folder)

bergsm, 2019-12-26:
Here's the latest log. https://pastebin.com/iPz8rnEt From what I can tell it looks like there's an issue importing the netifaces python module. Not sure if that's Cura or the plugin.

fieldOfView, 2019-12-26:
What version of Cura on what OS do you use?

fieldOfView, 2019-12-26:
Can you please try the official AppImage instead of whatever you use now? There are all sort of Python modules missing with the version you are using now, which also causes other plugins not to work.

bergsm, 2019-12-26:
This was Cura 4.4.0 on Arch Linux. Using the AppImage works. I think this issue may be similar to #98. I'll see about any missing dependencies due to my package manager. Either way it doesn't appear related to this repo :)