fieldOfView/Cura-OctoPrintPlugin #86: Weird issues with cura beta 3.5

lenne0815, 2018-09-20:
Depending on uninstalling / reinstalling the octoprint plugin and restoring my old profiles, cura beta 3.5 "forgets" my printers or crashes right at startup. It also crashes when everything is working "correctly" and i try to connect a printer.

I personally think its an error from my older printer profiles, but couldnt really pin it down.

[cura.log](https://github.com/fieldOfView/OctoPrintPlugin/files/2402685/cura.log)

fieldOfView, 2018-09-21:
Today a new version of the OctoPrintPlugin was added to Toolbox, which fixes two issues that you were running into. One was my fault, the other too. Could you test if OctoPrintPlugin 3.5.1 works for you?

lenne0815, 2018-09-21:
Great to hear, ill test it today !

lenne0815, 2018-09-21:
Tested, works now as expected ! Thank you !

closed 2018-09-21 by fieldOfView