fieldOfView/Cura-OctoPrintPlugin #87: Switch to GCodeWrite for serialization

Kriechi, 2018-10-05:
I recently got this issue on the Cure-DuetRRF plugin:
https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/14

See my recent change to use the GCodeWriter plugin:
https://github.com/Kriechi/Cura-DuetRRFPlugin/commit/ecba95c3f3e71f5d9387a2d9a192606ba21fba64#diff-4788f60738c07a5b6b9c54e7793555cf

Which I think would be suitable to replace this code:
https://github.com/fieldOfView/OctoPrintPlugin/blob/9d4fa5bfea77e8c649bba296b5b2732667b6ae5f/OctoPrintOutputDevice.py#L228-L233

Any thoughts?

Kriechi, 2018-10-05:
Looking through the open issues, I think this actually is a duplicate of #60!
Feel free to use my code snippet!

fieldOfView, 2018-10-05:
Nice trick! The only disadvantage I can think of is loosing control over which buildplate(s) is/are sent. But I can live with that.

Kriechi, 2018-10-06:
fixed by f1df408ebdadfbd21460833f2dfd0b15dca2d982

closed 2018-10-06 by Kriechi