fieldOfView/Cura-OctoPrintPlugin #89: No Button for Connection

scharrin2, 2018-10-19:
Hello,

I'm using Cura 3.5.1 on a Mac (Mojave). When I open the dialog "Manage Printers" there is no button for connection to OctoPrint.

I already uninstalled the PlugIn and reinstalled it. But the result is the same.

Best regards,
Christian

[image]
[image]

fieldOfView, 2018-10-19:
Please post your cura.log
The Cura GUI log file, located at

    %APPDATA%\cura\<Cura version>\cura.log (Windows), or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
    $USER/Library/Application Support/cura/<Cura version>/cura.log (OSX)
    $USER/.local/share/cura/<Cura version>/cura.log (Ubuntu/Linux)

scharrin2, 2018-10-19:
I deleted the cura.log and opened Cura again to create a new one.

[cura.log](https://github.com/fieldOfView/OctoPrintPlugin/files/2496613/cura.log)

fieldOfView, 2018-10-22:
The log says that the OctoPrintPlugin is disabled.
Go to Toolbox -> Browse packages... and on the "Installed" tab check that the "OctoPrint Connection" checkbox is not unchecked.