fieldOfView/Cura-OctoPrintPlugin #91: Time and percent complete are incorrect

critter42, 2018-11-08:
Every file I send to Octoprint has a time estimate that is double what Cura shows. Additionally, the model has completed printing, but Octoprint and Cura show it is 50% done, so I have to "abort print"  - it is like it is trying to print 2 copies all the time.

[image]

fieldOfView, 2018-11-08:
Does the same happen if you save the gcode, upload it to OctoPrint through the web interface and print it from there? If so, I don't think there's anything the OctoPrint Connection plugin can do about this; the plugin seems to show exactly what OctoPrint is showing.

fieldOfView, 2018-12-04:
I'm going to assume this is indeed a duplicate of #94. Feel free to reopen if the version that is currently published to the Marketplace in Cura does not fix it.

closed 2018-12-04 by fieldOfView