fieldOfView/Cura-OctoPrintPlugin #92: Plugin produces excess HTTP requests

ArmedSnail, 2018-11-13:
Cura 3.5.1 on Win10
OctoPrint 1.3.9 with OctoPi 0.15.1 on Raspbery Zero W

I recently had issues with Octoprint not beeing accessible when trying to slice and upload through Cura since the update to 3.5.1.

What I checked so far:
top on Raspberry shows me Octoprint running at ~2% Processor load when Idle and around 98% when Cura is running.
WireShark shows around ten HTTP GET going out when Cura is running so basically the Plugin is spamming my Octoprint until it is not reachable.
Cura Settings everything seems normal but in Printer selection when going into the "Connect to Octoprint" Menu the "checking the API Key" is flashing as if it constantly checks gets an OK and checks again (Cursor is not inside API field).

Also when Octoprint is not running the Plugin will spam "OctoPrint returned an error: Service not Running" or "OctoPrint returned an error: Bad Gateway" for every HTTP GET until Cura crashes
I will open a seperate issue for that if needed.

Wireshark dump of one second with Cura running where .2 is my Computer and .27 the Octoprint server:
`No.     Time           Source                Destination           Protocol Length Info
     15 0.005225       192.168.55.27         192.168.55.2          HTTP     529    HTTP/1.1 200 OK  (application/json)
     17 0.006644       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
     58 0.060315       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
     59 0.060331       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
     60 0.060350       192.168.55.2          192.168.55.27         HTTP     319    GET /api/settings HTTP/1.1 
     61 0.062111       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
     62 0.073090       192.168.55.27         192.168.55.2          HTTP     278    HTTP/1.1 409 CONFLICT  (text/html)
     68 0.147673       192.168.55.27         192.168.55.2          HTTP     704    HTTP/1.1 200 OK  (application/json)
     69 0.149190       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
     71 0.223367       192.168.55.27         192.168.55.2          HTTP     704    HTTP/1.1 200 OK  (application/json)
     72 0.225304       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
     74 0.280968       192.168.55.27         192.168.55.2          HTTP     278    HTTP/1.1 409 CONFLICT  (text/html)
     75 0.283957       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
     77 0.345310       192.168.55.27         192.168.55.2          HTTP     278    HTTP/1.1 409 CONFLICT  (text/html)
     78 0.350317       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
     80 0.422883       192.168.55.27         192.168.55.2          HTTP     704    HTTP/1.1 200 OK  (application/json)
     82 0.478415       192.168.55.27         192.168.55.2          HTTP     278    HTTP/1.1 409 CONFLICT  (text/html)
     83 0.480305       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
     85 0.576901       192.168.55.27         192.168.55.2          HTTP     704    HTTP/1.1 200 OK  (application/json)
     90 0.593719       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
     97 0.596204       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
     98 0.596271       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
    100 0.596824       192.168.55.2          192.168.55.27         HTTP     319    GET /api/settings HTTP/1.1 
    104 0.598853       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
    119 0.614007       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
    120 0.614007       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
    121 0.614018       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
    122 0.614037       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
    123 0.614076       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
    128 0.652129       192.168.55.27         192.168.55.2          HTTP     278    HTTP/1.1 409 CONFLICT  (text/html)
    143 0.688565       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
    145 0.738762       192.168.55.27         192.168.55.2          HTTP     704    HTTP/1.1 200 OK  (application/json)
    147 0.740587       192.168.55.2          192.168.55.27         HTTP     314    GET /api/job HTTP/1.1 
    149 0.801552       192.168.55.27         192.168.55.2          HTTP     278    HTTP/1.1 409 CONFLICT  (text/html)
    150 0.803594       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
    156 0.868483       192.168.55.27         192.168.55.2          HTTP     278    HTTP/1.1 409 CONFLICT  (text/html)
    157 0.870904       192.168.55.2          192.168.55.27         HTTP     318    GET /api/printer HTTP/1.1 
`

fieldOfView, 2018-11-14:
Am I correct that your OctoPrint instance is not being automatically listed in the Connect to OctoPrint dialog, and you needed to add it manually?

fieldOfView, 2018-11-14:
Do you have VMWare installed on the computer that you run Cura on? Or any other "non standard" network adapters? I am looking into why zeroconf.py raises an OSError, and found this:
https://github.com/jstasiak/python-zeroconf/issues/81

ArmedSnail, 2018-11-14:
> Am I correct that your OctoPrint instance is not being automatically listed in the Connect to OctoPrint dialog, and you needed to add it manually?

It was automatically listed, but it looked as if it was performing a check on the API Key constantly.
I removed it and tried to add it again but now it looks like the Octoprint instance is loosing Focus after 1 or 2 seconds so I am not able to put in an API key.

> Do you have VMWare installed on the computer that you run Cura on? Or any other "non standard" network adapters? I am looking into why zeroconf.py raises an OSError, and found this:
> [jstasiak/python-zeroconf#81](https://github.com/jstasiak/python-zeroconf/issues/81)

I have Oracle Virtual Box installed but without the Networking components.
What looks common is that i have assigned two IP Adresses on my Interface.
When I remove one of them everything works as expected.

So in fact it seems to be a zeroconf issue rather than a Plugin issue.

Thank you for the fast support :)

ArmedSnail, 2018-11-14:
> Am I correct that your OctoPrint instance is not being automatically listed in the Connect to OctoPrint dialog, and you needed to add it manually?

so after removing the second Adress from my LAN interface I know what you meant by "automatically listed".
It was never listed automatically, maybe because i added it myself some time ago. now it shows up as "octopi._octoprint._tcp.local

fieldOfView, 2018-11-14:
> So in fact it seems to be a zeroconf issue rather than a Plugin issue.

It actually looks like an issue that affects both zeroconf and the Qt network components. I will see if I can reproduce it by adding a second IP address to one of my interfaces, and once it is reproduced see if it can be worked around. I'mglad you found a workaround though.

fieldOfView, 2018-11-14:
@ArmedSnail could you test if this version of the plugin behaves better with network interfaces with multiple ipadresses?
http://files.fieldofview.com/cura/OctoPrintPlugin-v5.0.0-2018-11-14T21_29_30Z.curapackage
Just drop the downloaded file into a running Cura window, and restart Cura.

ArmedSnail, 2018-11-16:
I tested the File provided with multiple IP on CURA 3.5.1 and 3.6.0  and it looks OK.
It seems the detection of Octopi instance is not working anymore because it shows me my "old" configuration and not the one i created for testing purposes.

fieldOfView, 2018-11-16:
> It seems the detection of Octopi instance is not working

Please elaborate, I don't quite understand. What did you change, what do you expect to happen, what happens instead?

ArmedSnail, 2018-11-17:
Wehne first running into the Problem that the Plugin was spamming my Octopi, I removed the Octopi instance (called "octopi") from Cura for troubleshooting because I saw it was trying to verify the Key constantly. After that I was able to add it again (with name  "octopi1"), but because it lost Focus it was not possible to enter an API Key.
After removing one of the IP-Adresses from my Interface the problem with spamming and also loosing focus on the configuration was lost. Also it showed a ""octopi._octoprint._tcp.local" that I*ve never seen before. I deleted the "octopi1" and entered my API Key into the "octopi._octoprint._tcp.local" and used it for Printing.

After Importing the File you provided the automatically created "octopi._octoprint._tcp.local" was gone and the "octopi1" shows instead.

hope this helps.

fieldOfView, 2018-11-17:
Can you post a link to your cura.log?

ArmedSnail, 2018-11-17:
here is the Log from yesterday
[cura.log](https://github.com/fieldOfView/OctoPrintPlugin/files/2591625/cura.log)

fieldOfView, 2018-11-17:
Thanks. 

Silly me. Autodiscovery with zeroconf/bonjour still won't work if you have an interface with multiple IP addresses (because I did not change/fix zeroconf.py), but at least it won't crash Cura and your Pi anymore, right?

ArmedSnail, 2018-11-17:
I can confirm that the spamming stopped and the UI is fulli accessible again :)

fieldOfView, 2018-11-17:
Wohoo, thanks for confirming

closed 2018-11-17 by fieldOfView