fieldOfView/Cura-OctoPrintPlugin #93: Can't connect to OctoPrint via .local (possibly IPv6 issue?)

fieldOfView, 2018-11-16:
Reported here by @stevedc2:
https://github.com/Ultimaker/Cura/issues/4639

**Application Version**
3.5.1

**Octoprint Version**
Version 1.3.9 but other versions also exhibit the same behavior

**Platform**
Windows 10

**Printer**
CR10s & Mendel RepRap

**Steps to Reproduce**
Try to connect a printer to OctoPrint

**Actual Results**
I can't get Cura to connect to Octoprint, Cura continually sits "Checking the API key" and never returns.
I get this result whenever the printer IP/Hostname is set to a ".local" name, e.g. "cr10s.local". If I use an IPv4 address then everything seems OK.

If I ping the address "cr10s.local" then ping displays an IPv6 address, not an IPv4 address so I think there might be an issue with v6 addressing.

If I click "Open in browser" I correctly get a connection to Octoprint in my browser.

Another interesting observation is that if I go to the "Connect to Octoprint" dialog and paste the API key it sits "Checking the API". If I close the dialog and return to the normal Cura window then go back to manage printers and select "Connect Octoprint" again, I can no longer type anything in to the API key text box. Clicking in the box does not present a cursor and nothing I type appears. the only way to get back to being able to enter the key is to completely close Cura and start again. I get exactly the same result even if I add a completely new Octoprint instance.

Log file is attached

[cura.zip](https://github.com/Ultimaker/Cura/files/2521403/cura.zip)

fieldOfView, 2018-11-16:
Note that I still cannot reproduce this; .local and ipv6 work for me.

SteveDC2, 2018-11-16:
OK, did some more experimentation today and I THINK this is Windows being... well... Windows !!!

I tried connecting through a web browser with the local name (which was working fine previously, and was my solution when Cura wouldn't so I KNOW it worked) and Firefox wouldn't connect either today.

I tried a completely different machine and got the same result. 

So, I did a little research and came across the tip to flush the DNS cache on my Windows PC and BANG !!! Firefox and Cura can now connect again !!??

Weird thing is that the other PC can now connect through a browser to the .local name ???!!! Go figure.

Sometime I REALLY hate Windows, but most of the time just hate it !!!

I will keep an eye on this to see if Windows updates bork it, or if I can see any other pattern.