fieldOfView/Cura-OctoPrintPlugin #94: Duplicate print jobs through Octoprint plugin in Cura 3.6.0

jarrodasmith, 2018-11-17:
I want to make you aware of this issue:

https://github.com/Ultimaker/Cura/issues/4823

Not sure if it belongs there, or here...

Thanks in advance.

fieldOfView, 2018-11-18:
@cityeyes added:
> 
> I've been having the same issue. I initially thought it was some sort of user error, but it's only happened with the newest version of Cura and now I'm a bit relieved to see that other people are having the same problem.
> 
> I don't have Cura logs, but I may be able to grab Octoprint logs. I **do** have a timelapse of the print, so at least you can see the physical aspects of the issue.
> 
> You can see it [here](https://drive.google.com/open?id=1ciwOk5eUW0VxE1ElxwOsncEp49LvDg2b).
> 
> Long story short, the print finishes, sends the printer to ready state, waits a minute or two, then starts the second print, colliding with whatever has already been printed.

fieldOfView, 2018-11-18:
What version of the OctoPrint Connection plugin are you using? (See the "Installed" tab of the Marketplace)

Can you check the uploaded gcode file to see if the gcode is duplicated in the file?

cityeyes, 2018-11-18:
> What version of the OctoPrint Connection plugin are you using? (See the "Installed" tab of the Marketplace)
> 
> Can you check the uploaded gcode file to see if the gcode is duplicated in the file?

As far as I know, I was using the newest version of the plugin available at the time. Unfortunately, since I started having the issue, I deleted all old settings/user directory folders and reinstalled Cura fresh in an attempt to address another issue I was having, so maybe Jarrod will be able to chime in.

The gcode **does** look like it's sending duplicate instructions. I'm not super familiar with reading Gcode straight from the file, but it definitely looks like it goes through the shutdown procedure and then starts again. You can check the file that I looked at [here.](https://drive.google.com/open?id=14VijMiUf81uBDPhhMIb0FtAHSnkzjexI)

jarrodasmith, 2018-11-19:
I’m on 3.5.2 and Octoprint 1.3.9. I had another instance of the issue this morning.

jarrodasmith, 2018-11-19:
> What version of the OctoPrint Connection plugin are you using? (See the "Installed" tab of the Marketplace)
> 
> Can you check the uploaded gcode file to see if the gcode is duplicated in the file?

So, yes version 3.5.2, and I confirm that the gcode file that was uploaded to Octoprint appears to have 2 copies of the print.  I see 2 copies of the header:

;FLAVOR:Marlin
;TIME:5326
;Filament used: 5.26081m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.6.0

... and each one is followed by ~200k lines of M and G commands.

fieldOfView, 2018-11-19:
Thanks, it looks like the plugin duplicates the gcode. I am not sure why yet. This happens _sometimes_ for you, or is it fairly easily reproducible? Did it happens in Cura 3.5 or 3.4?

jarrodasmith, 2018-11-19:
> Thanks, it looks like the plugin duplicates the gcode. I am not sure why yet. This happens _sometimes_ for you, or is it fairly easily reproducible? Did it happens in Cura 3.5 or 3.4?

This happens only sometimes - but it isn't rare.  Maybe 10% of the time for me?  I have not been able to detect a pattern.  At first I thought maybe it was happening when I had more than one Cura instance open, connected to the same printer, but I just tested that and it seems to not happen in every instance in those cases.  So far it is still a mystery as to what exactly causes it.

Another variable to consider is that Fusion 360 is normally launching Cura for me.  Not sure if there could be an interaction there worth looking at, or not...

I have always used this plugin (since Cura 3.3?), and never saw this issue with previous versions of Cura.

cityeyes, 2018-11-19:
> Another variable to consider is that Fusion 360 is normally launching Cura for me. Not sure if there could be an interaction there worth looking at, or not...

I don't believe that's the trigger for this particular issue. I've used Fusion 360 in the past, but not recently. All of the times I've had a duplicate print from Cura, it was from a file opened directly in Cura.

Just grasping at straws, but I recall hitting "print" once or twice when my Octoprint instance wasn't connected, and then having to connect octoprint and hit print again to get it to work. I've done it a ton in the past, but maybe a recent change is causing the Gcode to be duplicated when I do that?

Outside of that, I can't think of anything I've done out of the ordinary that would cause it to duplicate.

My stats are about the same, maybe 10-20%. That said, I haven't been able to tinker with it much recently.

jarrodasmith, 2018-11-19:
> Just grasping at straws, but I recall hitting "print" once or twice when my Octoprint instance wasn't connected to my printer, and then having to connect octoprint and hit print again to get it to work. I've done it a ton in the past, but maybe a recent change is causing the Gcode to be duplicated when I do that?

I think this may be it.  Just tested the following:

1. Connect Cura to Octoprint instance that is *disconnected* from the serial interface of its printer.
2. Load STL
3. Slice
for (n=1; n<=N; n++){
   4. Choose "Print to Octoprint"
   5. Get the "Printer is disconnected, Queue the Job?" Dialog
   6. Dismiss the dialog by clicking the X (i.e. don't queue it).
}
7. Go to Octoprint web interface and connect it to the printer's serial interface.
8. Return to Cura and hit "Print to Octoprint"

The resulting uploaded file has N+1 copies of the Gcode.

fieldOfView, 2018-11-19:
Awesome, I can see how that would cause this duplication; this bug was indeed introduced in https://github.com/fieldOfView/OctoPrintPlugin/commit/f1df408ebdadfbd21460833f2dfd0b15dca2d982 but with this extra bit of info I spotted the actual mistake so I won't have to revert the whole thing. A fix is coming up.

fieldOfView, 2018-11-19:
The bug should be fixed with this version:
http://files.fieldofview.com/cura/OctoPrintPlugin-v5.0.0-2018-11-19T21_03_05Z.curapackage

Just drop the file into a running Cura window, and restart Cura.

jarrodasmith, 2018-11-19:
> The bug should be fixed with this version:
> http://files.fieldofview.com/cura/OctoPrintPlugin-v5.0.0-2018-11-19T21_03_05Z.curapackage
> 
> Just drop the file into a running Cura window, and restart Cura.

Confirmed fixed.  Thank you.

One note:  After updating in this way, Marketplace no longer shows this plugin under "installed".  Not sure how to get back onto the release version once it gets pushed out.

commit 6ba78a85 (2018-11-19): Prevent duplication of gcode when not queueing.

Fixes #94
closed 2018-11-21 by fieldOfView