fieldOfView/Cura-OctoPrintPlugin #95: Can't connect to octoprint in Cura 3.6.0

sotona-, 2018-11-20:
Hi,
Cura 3.6.0 win64 under Win7, plugin 3.5, OctoPrint 1.3.9 at omega 2 board.
I added octoprint instance. When I'm pressing to instance menu item in instances list in left part of window, elements with API key field and other options is showing in a right window part ans disappearing after 0.5-2 seconds. Also instance menu item became to inactive state. After fast clicking 5-10 times to menu item cura is crashing.
I tried to add octoprint instance with or without reverse proxy, via http or https, result is same. Also I saw nginx logs, plugin doesn't make any http requests.
Octoprint is working via browser and via android app with API key.

fieldOfView, 2018-11-21:
Could you test if this version of the plugin?
http://files.fieldofview.com/cura/OctoPrintPlugin-v5.0.0-2018-11-14T21_29_30Z.curapackage
Just drop the downloaded file into a running Cura window, and restart Cura.

fieldOfView, 2018-11-21:
Or better yet, test this version:
http://files.fieldofview.com/cura/OctoPrintPlugin-v5.0.0-2018-11-19T21_03_05Z.curapackage

fieldOfView, 2018-11-21:
This version is now available in the ~Toolbox~Marketplace

closed 2018-11-21 by fieldOfView

sotona-, 2018-11-21:
Thanks, working now.

chezhalcyonacres, 2019-01-14:
I have this same issue, Installed Cura yesterday, grabbed plugin from marketplace. Tried updating manually, no change. 
Windows 10/64 bit. Octoprint is on Ubuntu Mate distro. Printer is off.

rgriebl, 2019-01-23:
Please reopen this bug.
I also have the exact same issue on Win 10/64, newest plugin (3.5.3) from the Marketplace. The log is spammed with repeating: 
```
2019-01-01 15:48:29,036 - DEBUG - [zeroconf-ServiceBrowser__octoprint._tcp.local.] OctoPrintPlugin.OctoPrintOutputDevicePlugin._onServiceChanged [196]: Bonjour service added: octoprint.home._octoprint._tcp.local
2019-01-01 15:48:30,914 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevicePlugin._keepDiscoveryAlive [98]: Zeroconf discovery has died, restarting discovery of OctoPrint instances.
```
I tried to un-install Bonjour for Windows, tried to install a different version of Bonjour (2 instead of 3), but the error still stays the same.

fieldOfView, 2019-01-28:
@rgriebl Could you test http://files.fieldofview.com/cura/OctoPrintPlugin-v5.0.0-2019-01-28T14_45_08Z.curapackage ? Drop the file in a running Cura window and restart Cura.

rgriebl, 2019-01-30:
Sorry for the sparse feedback so far. Your new version is connecting perfectly fine to my OctoPrint instance now. Thanks a bunch for that.
There's always a 'but' though ... the webcam doesn't show. I saw a few bugreports about that, but since I never had the OctoPrint integration working before, I can just say that it doesn't work with this version. The log only shows 
``
2019-01-30 03:27:44,093 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [698]: Set OctoPrint camera url to http://octoprint.home:80/webcam/?action=stream
``
which is the correct streaming URL. I just wanted to mention it, since you are working on #101 in parallel.

fieldOfView, 2019-02-02:
Are you sure you ticked the box to display the camera image? The problem in #101 seems to have been somewhat unique in that it did show a camera still every 60 seconds or so, and it showed some specific output to the log about the MJPEG buffer exceeding reasonable size. Unless yours is showing the same issue (and you did check the "Show camera image" checkbox in the connect to OctoPrint dialog !), it is once more a different issue than #101

rgriebl, 2019-02-06:
I debugged this myself today and learned something new about QNetworkAccessManager after using it for over 10 years ... it _will_ send the port number in the HTTP "Host" header, whereas all browsers leave it out, if it's the default 80. That wouldn't be a problem in itself, but my haproxy on the octoprint machine was doing a hdr(host) instead of a hdr_dom(host) match and that trailing ":80" meant it never chose the correct backend. Doh...
Everything is working perfectly fine now after fixing the haproxy config. Sorry for the noise and thanks for your work on this plugin.

fieldOfView, 2019-02-06:
Thanks for the update, glad you have it working now.